bulletphysics/bullet3 #312: cmake + mingw make = internal compiler error: Segmentation fault

biller23, 2015-01-19:
Compilation give me a Segmentation fault
------------------------------------------------------
[ 25%] Building CXX object src/BulletCollision/CMakeFiles/BulletCollision.dir/CollisionDispatch/btHashedSimplePairCache.obj
[ 26%] Building CXX object src/BulletCollision/CMakeFiles/BulletCollision.dir/CollisionDispatch/btInternalEdgeUtility.obj
C:\bullet3-master\src\BulletCollision\CollisionDispatch\btInternalEdgeUtility.cpp: In member function 'virtual void btConnectivityProcessor::processTriangle(btVector3*, int, int)':
C:\bullet3-master\src\BulletCollision\CollisionDispatch\btInternalEdgeUtility.cpp:310:2: internal compiler error: Segmentation fault

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.

C:\bullet3-master\src\BulletCollision\CollisionDispatch\btInternalEdgeUtility.cpp:310:2: internal compiler error: Aborted

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
g++.exe: internal compiler error: Aborted (program cc1plus)
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
src\BulletCollision\CMakeFiles\BulletCollision.dir\build.make:726: recipe for target 'src/BulletCollision/CMakeFiles/BulletCollision.dir/CollisionDispatch/btInternalEdgeUtility.obj' failed
make[2]: *** [src/BulletCollision/CMakeFiles/BulletCollision.dir/CollisionDispatch/btInternalEdgeUtility.obj] Error 4
CMakeFiles\Makefile2:148: recipe for target 'src/BulletCollision/CMakeFiles/BulletCollision.dir/all' failed
make[1]: *** [src/BulletCollision/CMakeFiles/BulletCollision.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
------------------------------------------------------

Cmake configuration:

BUILD: all not selected
INSTALL: instal libs selected
INTERNAL: none selected
USE: none selected

Windows 7 x64
Cmake 3.1.0
MinGW (32bit)

erwincoumans, 2015-01-29:
As your report states, please report as a compiler bug:
Please submit a full bug report, with preprocessed source if appropriate. See http://gcc.gnu.org/bugs.html for instructions.

Can you check if the workaround in this closed issue works for you?
https://github.com/bulletphysics/bullet3/issues/39

biller23, 2015-01-30:
Yes I use latest repo.
I found a bug report of the same issue already open https://gcc.gnu.org/bugzilla/show_bug.cgi?id=63386

The compilation with -O1 flag works as suggested in that link, fail with -O2 or -O3.

erwincoumans, 2015-01-30:
They are asking for a preprocessed source version of that file in that issue. Perhaps you can ask them how to generate a preprocessed source version and help them fixing gcc/mingw?

biller23, 2015-01-30:
Unfortunately I cannot submit the files cause i have not an account and the account creation is not working. I have the file (.i) tho.

biller23, 2015-01-30:
GCC Bugzilla Admins block gmail addresses!?...
Anyway, now I submitted the file.

erwincoumans, 2015-02-03:
I cannot keep on adding workaround to Bullet for a broken compiler, I hope they fix gcc/mingw. Thanks for reporting to the right location. Let's close the issue.

closed 2015-02-03 by erwincoumans