bulletphysics/bullet3 #314: btConvexHullShape : constructor needs an offset?

skardous, 2015-01-21:
Hi,

I noticed an error in my application using  btConvexHullShape constructor with parameters. Perhaps I am mistaking, but it seems like constructor suggests that X, Y and Z coordinates are always int first positions of vertices data. That is not always true, for example you can have normals, uv before xyz values.

That is why i suggest to add an offset attribute to the constructor, like : 

```c++
    btConvexHullShape ::btConvexHullShape (const btScalar* points,int numPoints,int stride, int offset) :     btPolyhedralConvexAabbCachingShape ()
    {
     m_shapeType = CONVEX_HULL_SHAPE_PROXYTYPE;
     m_unscaledPoints.resize(numPoints);
   
    unsigned char* pointsAddress = (unsigned char*)points;
   
    for (int i=0;i<numPoints;i++)
    {
    btScalar* point = (btScalar*)pointsAddress;
    m_unscaledPoints[i] = btVector3(point[0 + offset], point[1 + offset], point[2 + offset]);
    pointsAddress += stride;
    }
   
    recalcLocalAabb();
   
    }
```

erwincoumans, 2015-01-21:
How about making 'points' point to your xyz value? It shouldn't matter in what order the xyz or uv values are.

skardous, 2015-01-22:
Thanks for your answer, finally it helped a lot. I added my offset to 'points' from start and everything went fine.

```c++
auto vertexBuffer = geometry->vertexBuffer("position"); 
    float* points = &vertexBuffer->data()[0];    
    int vertexBufferSize = vertexBuffer->numVertices();

    auto attr = vertexBuffer->attribute("position");
    auto offset = std::get<2>(*attr);
        
    auto size = vertexBuffer->vertexSize();
    

    auto btShape = std::shared_ptr<btConvexHullShape>(
        new btConvexHullShape(
            points + offset, 
            vertexBufferSize, 
            vertexBuffer->vertexSize() * sizeof(float)
        )
    );
```

closed 2015-01-22 by skardous