bulletphysics/bullet3 #329: -lboost_regex needed during linking

sjneph, 2015-02-21:
This is my first time building bullet3.  I'm using Fedora 20 with boost 1.54.0 installed through the package manager.

cd build3
./premake_linux64 gmake
cd gmake
make

Gives the same error in two build cases.
test_bullet_collision.make
Test_OpenCL_Bullet3.make

../../bin/libgtest_gmake_x64_release.a(gtest-all.o): In function `testing::internal::RE::~RE()':                                                                                                                   
gtest-all.cc:(.text+0x46ef): undefined reference to `regfreeA'
...

The fix was to include -lboost_regex to the LIBS variable to each of those two make files.

erwincoumans, 2015-02-21:
Bullet has no dependency on boost, this is a bug and any boost usage should be removed. If gtest really depends on boost, then we have to remove gtest or fix that.

sjneph, 2015-02-21:
Okay, that's a bad assumption on my part (boost dependency).  I will dig further into this.

erwincoumans, 2015-02-21:
It would be great if you can help sorting out the mess on Fedora with gtest. Can you check out how to link against POSIX regex on Fedora?

sjneph, 2015-02-21:
It looks like things compile fine. I was picking up boost's header for regex like a dummy.  Sorry for my confusion - I'll close this out.

closed 2015-02-21 by sjneph

tauruswang, 2019-02-12:
😊v  
-lboost_regex needed during linking    yes, gmake use the boost regex ???

erwincoumans, 2019-02-12:
Bullet doesn't use boost.