bulletphysics/bullet3 #332: Add implicit viscous joint damping

hsu, 2015-02-26:
Support viscous joint damping using a constraint on the joint's DOF with zero ERP and finite CFM.  Reference [KpKdToCFMERP or CFMERPToKpKd](https://bitbucket.org/osrf/gazebo/src/783d80d6567cebb1d350296145a87f311acfb38f/gazebo/physics/ode/ODEJoint.cc?at=default#cl-962)

scpeters, 2015-02-26:
One tricky aspect for us was switching the joint damping constraint on and off at joint limits (see [ODEJoint::ApplyImplicitStiffnessDamping](https://bitbucket.org/osrf/gazebo/src/783d80d6567cebb1d350296145a87f311acfb38f/gazebo/physics/ode/ODEJoint.cc#cl-842)).

erwincoumans, 2015-05-03:
I won't have time to implement the feature for Bullet 2.83. I wonder how your damping (and spring) compares with the btGeneric6DofSpring2Constraint version?

closed 2021-04-26 by erwincoumans