bulletphysics/bullet3 #333: Assert in multiply2_p8r

erwincoumans, 2015-02-27:
See description here:
http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=10413

I managed to avoid the assert by enabling the motors before the first step is made.

Apparently a btGeneric6DofConstraint has its m_numConstraintRows by default set to 0 but m_numConstraintRows is increased by 1 for every linear limit that needs to apply a force or angular limit that needs to apply a torque. For the last check to be positive, the motor needs to be enabled.
For me the assert was called when adding a second hinge constraint in the scene. So even though in other cases m_numConstraintRows of the btGeneric6DofConstraint was zero, it did not call the assert in those cases. So I'm not truly understanding what went wrong and why, but at least things are working :-)

imanbrouwer, 2015-03-02:
In case this is helpful, this is the code where I setup the joints. When I have 2 hinge joints enabled as in the code below, the code crashes as mentioned above

if( true )
    {
        btGeneric6DofConstraint * hinge = new btGeneric6DofConstraint(*m_bodies[PIVOT_PART], localA, false );
        // NOTE: 6DofConstraint only supports (currently) rotational motors!
        hinge->getRotationalLimitMotor(0)->m_loLimit = 1;
        hinge->getRotationalLimitMotor(0)->m_hiLimit = 0;
        hinge->getRotationalLimitMotor(0)->m_enableMotor = true;

        hinge->getRotationalLimitMotor(1)->m_loLimit = 1;
        hinge->getRotationalLimitMotor(1)->m_hiLimit = 0;
        hinge->getRotationalLimitMotor(1)->m_enableMotor = true;

        hinge->getRotationalLimitMotor(2)->m_loLimit = 1;
        hinge->getRotationalLimitMotor(2)->m_hiLimit = 0;
        hinge->getRotationalLimitMotor(2)->m_enableMotor = true;
        
	m_joints[PIVOT] = hinge;
	m_bulletWorld->addConstraint(hinge, true );
    }

    {
        btSliderConstraint * slider = new btSliderConstraint(*m_bodies[PIVOT_PART], *m_bodies[SHAFT], localA, shaftTipCenter, true);
		slider->setLowerLinLimit(1);
		slider->setUpperLinLimit(0);
        slider->setPoweredLinMotor( true );
        m_joints[SLIDER] = slider;
		m_bulletWorld->addConstraint(slider, true);
    }
    
    btVector3 zAxis(0,0,1);
    if( true )
    {        
        btHingeConstraint * hinge = new btHingeConstraint(*m_bodies[SHAFT], *m_bodies[UPPER_JAW], shaftUpperJawHingePos, upperJawHingePos, zAxis, zAxis );
        hinge->enableAngularMotor( true, 0, 0 );
        m_joints[UPPER_JAW_ROT] = hinge;
        m_bulletWorld->addConstraint(m_joints[UPPER_JAW_ROT], true );
    }
   
    if( true )
    {        
        btHingeConstraint * hinge = new btHingeConstraint(*m_bodies[SHAFT], *m_bodies[LOWER_JAW], shaftLowerJawHingePos, lowerJawHingePos, zAxis, zAxis );
        hinge->enableAngularMotor( true, 0, 0 );
        m_joints[LOWER_JAW_ROT] = hinge;
        m_bulletWorld->addConstraint(m_joints[LOWER_JAW_ROT], true );       
    }

erwincoumans, 2015-05-03:
The btGeneric6DofSpringConstraint will be deprecated and replaced by btGeneric6DofSpring2Constraint. Not sure if it will suffer the same issue. Thanks for the report!

simo-11, 2015-10-25:
I have similar issues with modified btGeneric6DofSpring2Constraint. In my case assert hits after I disable constraints due to breaking.
I could do some studying if someone has good ideas how this issue should be solved.

closed 2021-04-26 by erwincoumans