bulletphysics/bullet3 #334: btGeneric6DofSpring2Constraint blows up for kinematic rotate

erwincoumans, 2015-02-27:
See http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=10410

It also has a workaround

erwincoumans, 2015-03-01:
The mass (and inertia) of kinematic objects needs to be set to zero, otherwise all constraints don't work well. Need to double-check if this is really the issue. If the mass for kinematic objects is not set to zero, than the constraint solver writes back an incorrect velocity, which could cause the problem.

erwincoumans, 2015-03-08:
Reminder: check out PendulumDemo, send over email

simo-11, 2015-04-22:
I'm using btGeneric6DofSpring2Constraint in my project which tries to make bullet handle plasticity and it behaves reasonable if double precision and MLCP solver are used. 
In picture below SI solver is used and specimen jumps and breaks into two peaces before hammer hits it.
More details can be found in https://docs.google.com/document/d/1WA-BEf2ZuWSIU8XanfdIVvTK08Rk4Gy8H_Ll0X5bTsc/
[image]

erwincoumans, 2015-04-22:
Can you modify a default Bullet demo, and share the source code, so we can reproduce the problem easily? What are the masses of all objects involved, time step size and number of constraint solver iterations?

simo-11, 2015-04-22:
Source in in GitHub as fork for bullet and is based on Bullet demo.
Hammer is 20 kg and specimen two times 20 g. 
Timestep is automatically adjusted from 5 ms to 0.1 as hammer gets closer to specimen. Above 1.68 ms.
Number of solver iterations is not currently changed from default 10, but making it tunable goof be a good idea.
Appendix C: Building with Visual Studio Express 2013
Download source, run premake4 via vs2013.bat, open pbuild\vs2013\0_BulletPlasticity.sln, build Solution e.g. using DebugDouble
e.g. using GitHub PowerShell
D:\GitHub> git clone https://github.com/simo-11/bullet3 pbullet  
In cmd
D:\GitHub\pbullet\pbuild>vs2013.bat
D:\GitHub\pbullet\pbuild>vs2013\0_BulletPlasticity.sln

simo-11, 2015-04-24:
I made number of constraint solver iterations tunable and using values of 60 or higher produces reasonable results for my case.

simo-11, 2015-07-14:
After fixes by GaborPuhr solution is stable in my tests with single precision using 10 iterations.
https://github.com/bulletphysics/bullet3/commit/98e1334a64e0efd69afec20a58e3720d9f3ac87a

erwincoumans, 2015-07-14:
Those fixes are in latest Bullet github repository, so we can close this issue, right? If not, please create a new one. Thanks for the report and confirming it has been fixed.

closed 2015-07-14 by erwincoumans

simo-11, 2015-07-14:
Those fixes are in latest Bullet github repository, so issue can be closed.