bulletphysics/bullet3 #335: m_appliedImpulse in gContactProcessedCallback

velkyel, 2015-03-01:
there is a lot of question about zero value m_appliedImpulse in gContactProcessedCallback on forums (it's probably best value to start playing sounds etc.).

I think actual call gContactProcessedCallback in btPersistentManifold::refreshContactPoints() is too early for many cases.  Final right value m_appliedImpulse get manifold point in btSequentialImpulseConstraintSolver::solveGroupCacheFriendlyFinish().  So call gContactProcessedCallback should be somewhere after that.

I have modified my bullet fork witch is too different to make pull request with same patch.

erwincoumans, 2015-03-06:
thanks for the feedback, I'll consider it.

erwincoumans, 2015-05-03:
It won't make it for Bullet 2.83, but possibly in the next release (I really need to push out the 2.83 release, it has been delayed way too long)

closed 2021-04-26 by erwincoumans