bulletphysics/bullet3 #339: Create tags for previous releases

AMDmi3, 2015-03-12:
Could you please create tags for previous releases of bullet, at least 2.82? Since google code is closing, source consumers like packaging systems which still fetch from google code will switch to github and tag would be crucial for them until 2.83 is released. For instance, it'd be very useful to switch FreeBSD port of bullet to fetching from github. Older tags would also be nice for historical reasons.

scpeters, 2015-03-12:
+1 for [homebrew packaging](https://github.com/Homebrew/homebrew/blob/master/Library/Formula/bullet.rb#L5-L8) as well

scpeters, 2015-03-12:
I think 19f999ac087e68ffc2551ffb73e35e60271a0d27 should be the 2.82 tag

scpeters, 2015-03-12:
See [r2705](https://code.google.com/p/bullet/source/detail?r=2705) that declares the tag and [r2704](https://code.google.com/p/bullet/source/detail?r=2704) which matches 19f999a

AMDmi3, 2015-03-12:
> I think 19f999a should be the 2.82 tag
Agreed.

scpeters, 2015-03-25:
@erwincoumans I think you could just do the following:
~~~
git checkout 19f999a
git tag 2.82
git push --tags
~~~

scpeters, 2015-05-01:
I don't know when you did it, but I see it now. Thanks!

https://github.com/bulletphysics/bullet3/releases/tag/2.82

erwincoumans, 2015-05-03:
Yes, I added the 2.82 tag. Also, I will add the Bullet 2.83 tag very soon now.

AMDmi3, 2015-08-27:
Thanks! I think this may be closed now.

closed 2015-08-27 by AMDmi3