bulletphysics/bullet3 #341: Separate the manifold processing from collision processing

layzerar, 2015-03-17:
In some usage scenario, user doesn't need to know the exact contact point of the collision. Thus, it's happy if the user could got 50% performance improved through just rewrite the default near callback.

```C++
# collisionPair.m_algorithm->processCollision(&obj0Wrap,&obj1Wrap,dispatchInfo,&contactPointResult);
collisionPair.m_algorithm->processCollision(&obj0Wrap,&obj1Wrap,dispatchInfo);
collisionPair.m_algorithm->processManifoldResult(&obj0Wrap,&obj1Wrap,dispatchInfo,&contactPointResult);
```

A benchmark profile of bullet2:
[image]

velkyel, 2015-03-22:
```
collisionPair.m_algorithm->processManifoldResult(&obj0Wrap,&obj1Wrap,dispatchInfo,&contactPointResult);
````
...bullet api does not contains this function, what is that?

layzerar, 2015-03-23:
Oh, this is an improve suggestion. Processing manifold result is too heavy weight for some kind collision object just like sensor object who just want to know the overlapping objects.

So I wonder if we could separate the manifold result processing from collision processing.

erwincoumans, 2015-05-03:
I'll consider it for a release after Bullet 2.83, so let's keep this issue open. THanks for the request!

layzerar, 2015-05-05:
In my modified version, I add an array in `btManifoldResult` to store the contact points, then add a function `processContactPoints()` to process all contact points at once. And then I make `btManifoldResult`  instance shared by all of the collision algorithm in the same frame. I got a nice performance improvement.
```diff
 .../CollisionDispatch/btManifoldResult.cpp         | 168 +++++++++++----------
 .../CollisionDispatch/btManifoldResult.h           |  82 +++++++---
 2 files changed, 145 insertions(+), 105 deletions(-)

diff --git a/src/BulletCollision/CollisionDispatch/btManifoldResult.cpp b/src/BulletCollision/CollisionDispatch/btManifoldResult.cpp
index 4b2986a..1b2339c 100644
--- a/src/BulletCollision/CollisionDispatch/btManifoldResult.cpp
+++ b/src/BulletCollision/CollisionDispatch/btManifoldResult.cpp
@@ -59,96 +59,98 @@ btScalar	btManifoldResult::calculateCombinedRestitution(const btCollisionObject*
 }
 
 
-
-btManifoldResult::btManifoldResult(const btCollisionObjectWrapper* body0Wrap,const btCollisionObjectWrapper* body1Wrap)
-		:m_manifoldPtr(0),
-		m_body0Wrap(body0Wrap),
-		m_body1Wrap(body1Wrap)
-#ifdef DEBUG_PART_INDEX
-		,m_partId0(-1),
-	m_partId1(-1),
-	m_index0(-1),
-	m_index1(-1)
-#endif //DEBUG_PART_INDEX
-{
-}
-
-
-void btManifoldResult::addContactPoint(const btVector3& normalOnBInWorld,const btVector3& pointInWorld,btScalar depth)
+void btManifoldResult::processContactPoints(btPersistentManifold* manifoldPtr)
 {
-	btAssert(m_manifoldPtr);
+	btAssert(manifoldPtr);
 	//order in manifold needs to match
 
-	if (depth > m_manifoldPtr->getContactBreakingThreshold())
-//	if (depth > m_manifoldPtr->getContactProcessingThreshold())
-		return;
-
-	bool isSwapped = m_manifoldPtr->getBody0() != m_body0Wrap->getCollisionObject();
-
-	btVector3 pointA = pointInWorld + normalOnBInWorld * depth;
-
-	btVector3 localA;
-	btVector3 localB;
-	
-	if (isSwapped)
+	int i;
+	for (i = 0; i < m_numContactPoints; ++i)
 	{
-		localA = m_body1Wrap->getCollisionObject()->getWorldTransform().invXform(pointA );
-		localB = m_body0Wrap->getCollisionObject()->getWorldTransform().invXform(pointInWorld);
-	} else
-	{
-		localA = m_body0Wrap->getCollisionObject()->getWorldTransform().invXform(pointA );
-		localB = m_body1Wrap->getCollisionObject()->getWorldTransform().invXform(pointInWorld);
+		const btContactPoint& contactPoint = m_contactPointArray[i];
+		const btVector3& normalOnBInWorld = contactPoint.m_normalOnBInWorld;
+		const btVector3& pointInWorld = contactPoint.m_pointInWorld;
+		const btScalar depth = contactPoint.m_depth;
+
+		// if (depth > m_manifoldPtr->getContactProcessingThreshold())
+		if (depth > manifoldPtr->getContactBreakingThreshold())
+			continue;
+
+		bool isSwapped = manifoldPtr->getBody0() != m_body0Wrap->getCollisionObject();
+
+		btVector3 pointA = pointInWorld + normalOnBInWorld * depth;
+
[... 228 lines cut ...]
-	SIMD_FORCE_INLINE	void refreshContactPoints()
+	SIMD_FORCE_INLINE void refreshContactPoints(btPersistentManifold* manifoldPtr)
 	{
-		btAssert(m_manifoldPtr);
-		if (!m_manifoldPtr->getNumContacts())
+		btAssert(manifoldPtr);
+		if (!manifoldPtr->getNumContacts())
 			return;
 
-		bool isSwapped = m_manifoldPtr->getBody0() != m_body0Wrap->getCollisionObject();
+		bool isSwapped = manifoldPtr->getBody0() != m_body0Wrap->getCollisionObject();
 
 		if (isSwapped)
 		{
-			m_manifoldPtr->refreshContactPoints(m_body1Wrap->getCollisionObject()->getWorldTransform(),m_body0Wrap->getCollisionObject()->getWorldTransform());
+			manifoldPtr->refreshContactPoints(m_body1Wrap->getCollisionObject()->getWorldTransform(), m_body0Wrap->getCollisionObject()->getWorldTransform());
 		} else
 		{
-			m_manifoldPtr->refreshContactPoints(m_body0Wrap->getCollisionObject()->getWorldTransform(),m_body1Wrap->getCollisionObject()->getWorldTransform());
+			manifoldPtr->refreshContactPoints(m_body0Wrap->getCollisionObject()->getWorldTransform(), m_body1Wrap->getCollisionObject()->getWorldTransform());
 		}
 	}
 

```

closed 2021-04-26 by erwincoumans