bulletphysics/bullet3 #347: Assert/crash in btDbvt::collideOCL

antarespilot, 2015-03-28:
We've stumbled upon a bug in btDbvt::collideOCL. We're using the current master branch of bullet3 (currently at 29949b63ccd1ba7720fc9f4d09c5184d5e656ff1)

If running Bullet in debug mode (or at least when having DEBUG defined), we hit an assertion in btAlignedObjectArray:

```
/LinearMath/btAlignedObjectArray.h:178: T& btAlignedObjectArray<T>::operator[](int) [with T = int]: Assertion `n<size()' failed.
```

The assert occurs at line 1197 in btDbvt.h, at the memmove operation. The size of "stack" is 2, and the value of "j" is 1, meaning that when calling 

```c++
memmove(&stack[j+1],&stack[j],sizeof(int)*(stack.size()-j-1));
```

the first argument will go beyond the current size of "stack", thus triggering the assert. When running a release build, there is no assert, and everything seems to be working fine. Still there is something there that could write beyond the size of the "stack" array, which could cause other weird issues. 

Below is a complete "main.cpp" that reproduces the issue. This uses Bullet in double-precision mode. It does view frustum culling on a real-earth-sized world and is extracted from a larger application. 

I have not tried to reproduce this in single-precision-mode, but it seems like this issue has been around for quite a while, see http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?t=9112
This guy does not mention anything about double-precision. 

I'm not sure what has to be done to fix this, as I am not familiar with exactly how the collideOCL function works. Maybe it's as easy as an additional check on the size of "stack"? 

```c++
#include <iostream>
#include <BulletCollision/BroadphaseCollision/btDbvt.h>

using namespace std;

#ifndef M_PI
#define M_PI		3.14159265358979323846
#endif

const double deg_to_rad = M_PI/180.0;
const double earthRadius    = 6378000.0; ///< Radius for spherical earth calculations.

/// Convert from spherical latitude/longitude to geocentric XYZ coordinates.
inline btVector3 geoToXYZ(double lat, double lon, double alt)
{
   btVector3 ret;

   double phi = lat * deg_to_rad;
   double lam = lon * deg_to_rad;
   double r   = earthRadius + alt;

   double sin_phi = sin(phi);
   double cos_phi = cos(phi);
   double sin_lam = sin(lam);
   double cos_lam = cos(lam);

   ret.setX(r * cos_phi * cos_lam);
   ret.setY(r * cos_phi * sin_lam);
   ret.setZ(r * sin_phi);

   return ret;
}

//===========================================================================

void calcViewCollisionFrustum(const btVector3& eye_pos,
                              const btVector3& view_dir,
                              btScalar* plane_offsets,
                              btVector3* plane_normals)
{
    double hfovx = 45.0;
    double hfovy = 25.0;
    double farplane = 20000000;

    btVector3 up = eye_pos;
    up.normalize();

    btVector3 side = view_dir.cross(up);
    side.normalize();

    btMatrix3x3 view_to_earth(view_dir.x(), side.x(), up.x(),
                              view_dir.y(), side.y(), up.y(),
                              view_dir.z(), side.z(), up.z());
    view_to_earth = view_to_earth.transpose();

    // just to verify that view_dir is correct
//    btVector3 t =  btVector3(1.0,0.0,0.0)*view_to_earth;

    btVector3 c00,c10,c01,c11;
    c00 = btVector3(1.0,0.0,0.0);
    c00 = c00.rotate(btVector3(0.0,0.0, 1.0), -hfovx*deg_to_rad);
    c00 = c00.rotate(btVector3(0.0,-1.0,0.0),  hfovy*deg_to_rad);

    c01 = btVector3(1.0,0.0,0.0);
    c01 = c01.rotate(btVector3(0.0,0.0, 1.0), hfovx*deg_to_rad);
    c01 = c01.rotate(btVector3(0.0,-1.0,0.0), hfovy*deg_to_rad);

    c10 = btVector3(1.0,0.0,0.0);
    c10 = c10.rotate(btVector3(0.0,0.0, 1.0), -hfovx*deg_to_rad);
    c10 = c10.rotate(btVector3(0.0,-1.0,0.0), -hfovy*deg_to_rad);

    c11 = btVector3(1.0,0.0,0.0);
    c11 = c11.rotate(btVector3(0.0,0.0, 1.0),  hfovx*deg_to_rad);
    c11 = c11.rotate(btVector3(0.0,-1.0,0.0), -hfovy*deg_to_rad);

    c00 = c00 * view_to_earth;
    c10 = c10 * view_to_earth;
    c01 = c01 * view_to_earth;
    c11 = c11 * view_to_earth;

    plane_normals[0] = c01.cross(c00).normalized();
    plane_normals[1] = c10.cross(c11).normalized();
    plane_normals[2] = c00.cross(c10).normalized();
    plane_normals[3] = c11.cross(c01).normalized();
    plane_normals[4] = -view_dir;

    plane_offsets[0] = -eye_pos.dot(plane_normals[0]);
    plane_offsets[1] = -eye_pos.dot(plane_normals[1]);
    plane_offsets[2] = -eye_pos.dot(plane_normals[2]);
    plane_offsets[3] = -eye_pos.dot(plane_normals[3]);
    plane_offsets[4] = -(eye_pos + view_dir*farplane).dot(plane_normals[4]);
}

//===========================================================================

struct DbvtRenderer : btDbvt::ICollide
{
    void Process(const btDbvtNode* n,btScalar) { Process(n); }
    void Process(const btDbvtNode* node)
    {
        const char* name = (const char*)node->data;
        cerr << "hit node: " << name << endl;
    }
};

//===========================================================================

btDbvtVolume createCollisionBox(const btVector3& p, double r = 1.0)
{
    btVector3 bb_min(p.x() - r,  p.y() - r, p.z() - r);
    btVector3 bb_max(p.x() + r,  p.y() + r, p.z() + r);
    return btDbvtVolume::FromMM(bb_min,bb_max);
}

//===========================================================================

int main(int argc, char *argv[])
{
    DbvtRenderer dbvt_renderer;

    btVector3 eye_pos  = geoToXYZ(60.0, 10.0, 0.0);
    btVector3 box1_pos = geoToXYZ(60.01, 10.0, 0.0);
    btVector3 box2_pos = geoToXYZ(60.02, 10.0, 0.0);
    btVector3 box3_pos = geoToXYZ(60.03, 10.01, 0.0);
    btVector3 box4_pos = geoToXYZ(59.0, 10.0, 0.0);
    btVector3 view_dir = (box1_pos - eye_pos).normalized();

    btDbvt* dbvt = new btDbvt;
    btDbvtNode* n1 = dbvt->insert(createCollisionBox(box1_pos), (void*)"node 1");
    btDbvtNode* n2 = dbvt->insert(createCollisionBox(box2_pos), (void*)"node 2");
    btDbvtNode* n3 = dbvt->insert(createCollisionBox(box3_pos), (void*)"node 3");
    btDbvtNode* n4 = dbvt->insert(createCollisionBox(box4_pos), (void*)"outside node");

    btVector3 plane_normals[5];
    btScalar  plane_offsets[5];

    calcViewCollisionFrustum(eye_pos, -view_dir, plane_offsets, plane_normals);

    cerr << "Unsorted: " << endl;
    btDbvt::collideKDOP(dbvt->m_root, plane_normals, plane_offsets, 5, dbvt_renderer);

    cerr << "Sorted: " << endl;
    btVector3 sortaxis = plane_normals[4];
    btDbvt::collideOCL(dbvt->m_root, plane_normals, plane_offsets, sortaxis, 5, dbvt_renderer);
}
```

erwincoumans, 2015-05-03:
It looks like an issue of the use of memmove, should be fixed in upcoming pull request/Bullet 2.83
https://github.com/bulletphysics/bullet3/commit/7a0288bf56b5ae7c7bb1f7de99fccc397bec5f47
Thanks for the report!

 #if DBVT_USE_MEMMOVE
-							memmove(&stack[j+1],&stack[j],sizeof(int)*(stack.size()-j-1));
+							memmove(&stack[j],&stack[j-1],sizeof(int)*(stack.size()-j-1));
 #else
-							for(int k=stack.size()-1;k>j;--k) stack[k]=stack[k-1];
+							for(int k=stack.size()-1;k>j;--k) 
+							{
+								stack[k]=stack[k-1];
+							}
 #endif

antarespilot, 2015-05-04:
Hi Erwin,

i tried it out here using the fix. There is no crash anymore, but the frustum culling is still broken using memmove. Basically items inside the view frustum flickers on and off seemingly at random and at a high frequency. So we're disabling memmove again for now, the alternative for-loop implementation works perfectly. 

So I think the issue needs to be re-opened, there is still something wrong there. 

For the 2.83-release maybe a quick fix is to just disable memmove for now.

erwincoumans, 2015-05-04:
ok, re-opened the issue. The Bullet 2.83 release is already tagged, so a fix/workaround will be for 2.84. It should be trivial for you to workaround, right?

antarespilot, 2015-05-04:
No problem :-)

erwincoumans, 2015-05-04:
I have never used memmove, and wonder if the memmove is wrongly used, or if the memmove on your platform is wrongly implemented. memmove seems to deal with overlapping source/destination. Do you mind figuring out memmove, and how it compares with the working C reference using a for loop?

antarespilot, 2015-10-21:
I started looking at this again now, hope to make some kind of fix in the end.

antarespilot, 2015-10-21:
I'll make a new pull request tomorrow.

antarespilot, 2015-10-22:
Regarding comparing the isolated performance of the for-loop versus the memmove, I tried the Google benchmark library (https://github.com/google/benchmark).

Here is the complete code I used:

```c++
#include <benchmark/benchmark_api.h>
#include <vector>
#include <cstring>

int stacksize = 20000;
int j = 100;

void BM_forloop(benchmark::State& state) {

    std::vector<int> stack;
    stack.resize(stacksize);

    while (state.KeepRunning()) {
        for(int k=stack.size(); k>j; --k) {
            stack[k] = stack[k-1];
        }
    }
}

BENCHMARK(BM_forloop);

void BM_memmove(benchmark::State& state) {

    std::vector<int> stack;
    stack.resize(stacksize);

    while (state.KeepRunning()) {
        int num_items_to_move = stack.size()-1-j;
        if(num_items_to_move > 0)
            memmove(&stack[j+1],&stack[j],sizeof(int)*num_items_to_move);
    }
}

BENCHMARK(BM_memmove);

void BM_unsafememmove(benchmark::State& state) {

    std::vector<int> stack;
    stack.resize(stacksize);

    while (state.KeepRunning()) {
        memmove(&stack[j+1],&stack[j],sizeof(int)*(stack.size()-1-j));
    }
}

BENCHMARK(BM_unsafememmove);

BENCHMARK_MAIN()
```

The result output on my Linux machine looks like this:

```
Run on (8 X 2261 MHz CPU s)
2015-10-22 21:46:07
Benchmark          Time(ns)    CPU(ns) Iterations
-------------------------------------------------
BM_forloop            12736      12667      51471                                 
BM_memmove             3083       3084     224359                                 
BM_unsafememmove       3060       3062     227273  
```

First time I tried the benchmark library so I'm no expert at it. Found it after watching this here,
https://youtu.be/nXaxk27zwlk

antarespilot, 2015-11-12:
Fixed in master

closed 2015-11-12 by antarespilot