bulletphysics/bullet3 #355: Question: array of btTransofrm for adding child shapes

Ehsanizadi, 2015-04-13:
Hi,

I have couple of box shapes and I want to add them to a compound shape.
For the local transform declaration, it is more convenient in my case to define an array of btTransforms, like this:
```

btTransform    localTransofrm[8];
localTransform[0].setOrigin(btVector3(bla, bla, bla);
.
.
.
localTransofrm[8].setOrigin(btVector3(bla, bla, bla));

for (int i=0; i<9; i++){
    compoundObjectShape->addChildShape( localTransform[i], otherShape[i]);
}
```
This compiles well, however, I get segmentation fault when running the code.

But when I declare the localTransform (s) separately, it works well. 

My question is that is there any sort of limitations for definition of btTransform (s) in the form of arrays?
Here, I just have 9 childShapes which is not hard to separately define them, but what about the case of many childShapes??
(Or perhaps I am mistaking about something?)

Thank you.

scpeters, 2015-04-13:
Your array is declared of length 8, but you are accessing the 9th element. Is that the problem?
~~~
// Try this
btTransform    localTransform[9];
~~~

Ehsanizadi, 2015-04-13:
I don't think so, since the array index starts from zero, so, it will be total number of 9 arrays of btTransfroms.

scpeters, 2015-04-13:
http://www.tutorialspoint.com/cprogramming/c_arrays.htm

erwincoumans, 2015-04-13:
It is better to store btTransform object in a btAlignedObjectArray, to make sure it is memory-aligned (SIMD requires it, otherwise you can get crashes). So use 
btAlignedObjectArray<btTransform> localTransform;
localTransform.resize(8);

and do as you did before.

scpeters, 2015-04-13:
The seg-faults occur because of accessing beyond the end of the array.
~~~
btTransform    localTransform[8];
// This is incorrect
localTransform[8].setOrigin(btVector3(1, 2, 3));
~~~

Ehsanizadi, 2015-04-13:
Ok, 
Thank you Scpeters and Erwin, for the help.

These solved my problem.

Ehsanizadi, 2015-04-13:
Just adding a comment for prospective readers:
even if I use
```
btTransform    localTransform[9];
```
I still get segmentation fault.
The following is a part of output Valgrind (as I use it for profiling) shows:
```
.
.
.
==8926== Use of uninitialised value of size 8
==8926==    at 0x404175: btTransform::setOrigin(btVector3 const&) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==8926==    by 0x402075: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==8926== 
==8926== Invalid write of size 8
==8926==    at 0x404175: btTransform::setOrigin(btVector3 const&) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==8926==    by 0x402075: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==8926==  Address 0xe6854bb7 is not stack'd, malloc'd or (recently) free'd
==8926== 
==8926== 
==8926== Process terminating with default action of signal 11 (SIGSEGV)
==8926==  Access not within mapped region at address 0xE6854BB7
==8926==    at 0x404175: btTransform::setOrigin(btVector3 const&) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==8926==    by 0x402075: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
.
.
.
```

erwincoumans, 2015-04-13:
So the `btAlignedObjectArray<btTransform> localTransform; localTransform.resize(8);' fixed the issue, because of alignment, right?
Another workaround would be to disable SIMD all together, see 'LinearMath/btScalar.h' for how to do that, dependent on your platform.

Ehsanizadi, 2015-04-13:
Oop!
I thought it will solve my problem , because I see the usage of btAlignedObjectArray in demos.
However, it complies  and runs well in the demos. (thus, I reopen the issue, sorry for closing it beforehand)

But in my code even usage of btAlignedObjectArray did not work! 
Also I changed the size of that from 8 to 9, but same segfault error.

I am working on Ubuntu 14.04 (64bit)

I will be trying disabling SIMD, then I will post the outcome...

erwincoumans, 2015-04-13:
'didn't work' is not enough of an explanation, it should work. If not, provide the full compiler/linker error.

Make sure to include the header file ,`LinearMath/btAlignedObjectArray.h`

Ehsanizadi, 2015-04-13:
Well, the problem was definition of child collision shapes as array. (I defined local transforms and collision shapes of children both as array).
I used btAlignedObjectArray for both of them, now there is no segfault. Compiles well and runs well.

However, in the profiling using valgrind there seems to be some errors as below. I just post them to report, I am not sure if they really troublesome or not:
```
 .
.
.
==11818== Conditional jump or move depends on uninitialised value(s)
==11818==    at 0x462789: btDbvt::update(btDbvtNode*, btDbvtAabbMm&, btVector3 const&, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x429921: btDbvtBroadphase::setAabb(btBroadphaseProxy*, btVector3 const&, btVector3 const&, btDispatcher*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433A8F: btCollisionWorld::updateSingleAabb(btCollisionObject*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433C65: btCollisionWorld::updateAabbs() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x431F91: btCollisionWorld::performDiscreteCollisionDetection() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x40A656: btDiscreteDynamicsWorld::internalSingleStepSimulation(float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x406A09: btDiscreteDynamicsWorld::stepSimulation(float, int, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x402A5F: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818== 
==11818== Conditional jump or move depends on uninitialised value(s)
==11818==    at 0x462798: btDbvt::update(btDbvtNode*, btDbvtAabbMm&, btVector3 const&, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x429921: btDbvtBroadphase::setAabb(btBroadphaseProxy*, btVector3 const&, btVector3 const&, btDispatcher*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433A8F: btCollisionWorld::updateSingleAabb(btCollisionObject*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433C65: btCollisionWorld::updateAabbs() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x431F91: btCollisionWorld::performDiscreteCollisionDetection() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x40A656: btDiscreteDynamicsWorld::internalSingleStepSimulation(float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x406A09: btDiscreteDynamicsWorld::stepSimulation(float, int, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x402A5F: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818== 
==11818== Conditional jump or move depends on uninitialised value(s)
==11818==    at 0x4627A7: btDbvt::update(btDbvtNode*, btDbvtAabbMm&, btVector3 const&, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x429921: btDbvtBroadphase::setAabb(btBroadphaseProxy*, btVector3 const&, btVector3 const&, btDispatcher*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433A8F: btCollisionWorld::updateSingleAabb(btCollisionObject*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433C65: btCollisionWorld::updateAabbs() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x431F91: btCollisionWorld::performDiscreteCollisionDetection() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x40A656: btDiscreteDynamicsWorld::internalSingleStepSimulation(float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x406A09: btDiscreteDynamicsWorld::stepSimulation(float, int, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x402A5F: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818== 
==11818== Conditional jump or move depends on uninitialised value(s)
==11818==    at 0x4627B8: btDbvt::update(btDbvtNode*, btDbvtAabbMm&, btVector3 const&, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x429921: btDbvtBroadphase::setAabb(btBroadphaseProxy*, btVector3 const&, btVector3 const&, btDispatcher*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433A8F: btCollisionWorld::updateSingleAabb(btCollisionObject*) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x433C65: btCollisionWorld::updateAabbs() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x431F91: btCollisionWorld::performDiscreteCollisionDetection() (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x40A656: btDiscreteDynamicsWorld::internalSingleStepSimulation(float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x406A09: btDiscreteDynamicsWorld::stepSimulation(float, int, float) (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818==    by 0x402A5F: main (in /home/ehsan/Desktop/Bullet-Examples/directShearTest/output.out)
==11818== 
[... cut ...]
==11818== 
==11818== 
==11818== HEAP SUMMARY:
==11818==     in use at exit: 0 bytes in 0 blocks
==11818==   total heap usage: 160,116 allocs, 160,116 frees, 335,220,876 bytes allocated
==11818== 
==11818== All heap blocks were freed -- no leaks are possible
==11818== 
==11818== For counts of detected and suppressed errors, rerun with: -v
==11818== Use --track-origins=yes to see where uninitialised values come from
==11818== ERROR SUMMARY: 10000000 errors from 93 contexts (suppressed: 0 from 0)
```

erwincoumans, 2015-04-13:
Best to close now. For further issue it is best to use the forums at http://bulletphysics.org, unless you are very sure there is a bug in Bullet itself. Thanks!

closed 2015-04-13 by erwincoumans