bulletphysics/bullet3 #357: Fix GyroscopicDemo

benelot, 2015-04-15:
It is pretty easy actually!

It sets the flag on body to:
body->setFlags(BT_ENABLE_GYROPSCOPIC_FORCE);

It seems that the flag name has changed and is now:
body->setFlags(BT_ENABLE_GYROSCOPIC_FORCE_EXPLICIT);

Can anybody that understands the meaning of the flag confirm that this is the fix?

When I set the new flag, the code compiled properly and the demo works as I would expect it.

Thanks!

scpeters, 2015-04-15:
Pull requests #348 and #350 would fix this, but they haven't been merged yet.

erwincoumans, 2015-04-15:
ok, I merged the pull request (somehow I thought it was already merged, but it wasn't).
Note you should not use the 'explicit' version, but the 'implicit' one, to avoid adding energy/blowing up your simulation. BT_ENABLE_GYROPSCOPIC_FORCE maps to the implicit version now.

closed 2015-04-22 by erwincoumans