bulletphysics/bullet3 #359: Solving constraints becomes very slow if the number of islands is large

cameronwhite, 2015-04-17:
For every island, `InplaceSolverIslandCallback::processIsland()` iterates through the entire list of constraints to find the constraints that belong to that island. Since the constraints are already sorted by island id, it would be much faster to iterate through `m_sortedConstraints` once in `InplaceSolverIslandCallback::setup()` and precompute the index into `m_sortedConstraints` where each island's constraints begin.

erwincoumans, 2015-04-30:
sure. Do you mind preparing a patch for this?

benelot, 2016-05-03:
Is this issue fixed? It seems to be referenced in commit. If it is done, closing it would clean up the tracker.

cameronwhite, 2016-05-03:
https://github.com/bulletphysics/bullet3/pull/386 is the active pull request

benelot, 2016-05-03:
Ok, perfect, I am just trying to cleanup older issues. So this one is obviously not done yet. Thank you for linking it here!

erwincoumans, 2017-09-14:
Unfortunately I don't see the benefits, neither some good benchmark that shows the benefits. I ran the Ragdolls benchmark, with and without the patch, and performance is about the same (even slightly slower). See comment in the closed [pull request](https://github.com/bulletphysics/bullet3/pull/386). Let's close it for now.

closed 2017-09-14 by erwincoumans