bulletphysics/bullet3 #364: API changes in master branch

scpeters, 2015-05-01:
As an FYI to other users, the recently merged #363 made some API changes.

In f4b0cc85cdf5a6f8292bbd1b50b57d3fb3b02d0d for example, the `btHinge2Constraint` class is changed to derive from `btGeneric6DofSpring2Constraint` instead of `btGeneric6DofSpringConstraint`. The main issue is that now the `btHinge2Constraint::getRotationalLimitMotor` function returns type `btRotationalLimitMotor2` which does not inherit from `btRotationalLimitMotor`. We can work around this with `ifdef`'s if backward compatibility can't be achieved.

erwincoumans, 2015-05-01:
The btGeneric6DofSpringConstraint is going to be deprecated and removed in the future. Is there a particular feature/API thing missing from the new btGeneric6DofSpring2Constraint and/or btRotationalLimitMotor2? Thanks for the feedback!

scpeters, 2015-05-01:
I haven't checked for missing features. I was just noticing a failed gazebo build, so we'll probably have to use ifdef's until bullet 2.83 is available on Ubuntu platforms.

bchretien, 2015-05-12:
:+1: Gazebo compilation is currently broken on Arch Linux (Bullet 2.83) because of this.

closed 2021-04-26 by erwincoumans