bulletphysics/bullet3 #370: 2.83 tag changed

scpeters, 2015-05-05:
The 2.83 tag was created on Sunday night, but then it was modified on Monday. It's my understanding that tags are supposed to be immutable references to specific points in the code and should not be changed. It wreaks havoc on downstream users who package the code. I would recommend using patch release numbering (2.83.1) rather than modify the original tag.

I think @sherm1 did the same thing recently with simbody, so you're not alone. But I think I complained to him too.

It's done now, probably no need to revert it, but I would recommend patch release numbering for the future.

erwincoumans, 2015-05-05:
True, I thought no-one would notice this on a Sunday. Hey, stop working in the weekend Stephen :-)
Fair enough, I'll use patch release next time.

closed 2015-05-05 by erwincoumans

scpeters, 2015-05-05:
:)

sherm1, 2015-05-05:
> I think @sherm1 did the same thing recently with simbody, so you're not alone. But I think I complained to him too.

Guilty, but repentant and reformed!