bulletphysics/bullet3 #371: 2.83 build fails on Mountain Lion unless BUILD_BULLET2_DEMOS=OFF

scpeters, 2015-05-06:
There is already some logic in the root CMakeLists.txt to disable building the Bullet3 demos on OS X 10.8 (Mountain Lion) since it doesn't have OpenGL3. This logic should be extended to the bullet2 demos as well, since they are now using OpenGL as well. Otherwise I get the following error:
~~~
/tmp/bullet20150505-6635-1kir2za/bullet3-2.83/examples/OpenGLWindow/GLInstancingRenderer.cpp:1487:3: error: use of undeclared identifier 'glVertexAttribDivisor'; did you mean 'glVertexAttrib4iv'?
                glVertexAttribDivisor(0, 0);
                ^~~~~~~~~~~~~~~~~~~~~
                glVertexAttrib4iv
/System/Library/Frameworks/OpenGL.framework/Headers/gl3.h:1588:21: note: 'glVertexAttrib4iv' declared here
GLAPI void APIENTRY glVertexAttrib4iv (GLuint index, const GLint *v);
                    ^
/tmp/bullet20150505-6635-1kir2za/bullet3-2.83/examples/OpenGLWindow/GLInstancingRenderer.cpp:1488:3: error: use of undeclared identifier 'glVertexAttribDivisor'; did you mean 'glVertexAttrib4iv'?
                glVertexAttribDivisor(1, 1);
                ^~~~~~~~~~~~~~~~~~~~~
                glVertexAttrib4iv
...
11 errors generated.
make[2]: *** [examples/OpenGLWindow/CMakeFiles/OpenGLWindow.dir/GLInstancingRenderer.o] Error 1
make[1]: *** [examples/OpenGLWindow/CMakeFiles/OpenGLWindow.dir/all] Error 2
make: *** [all] Error 2
~~~
I discovered this while working on a [homebrew formula for 2.83](https://github.com/Homebrew/homebrew/pull/39366). I can make a workaround for the formula, but I thought I would report it upstream as well.

erwincoumans, 2015-05-06:
The Bullet 3 and Bullet 2 demos are merged into App_ExampleBrowser. So we need to fix compilation, using some 'USE_OPENGL2' define, so on older platforms it will only compile the OpenGL2 fallback. I'll see what I can do.

erwincoumans, 2015-05-06:
temp workaround, disabling all examples, is in this pull request. I'll see if I can re-enable the examples with only compiling OpenGL2 parts, using ifdefs
https://github.com/bulletphysics/bullet3/pull/373

erwincoumans, 2015-05-06:
it also includes some other essential bug fix. What do you suggest for a new patch release?
Name it 2.83.1 and delete the old 2.83 tag? I rather not have people use the broken tag.

erwincoumans, 2015-05-06:
By the way, what build system are you using for 'homebrew'? Who is using the --with-demos option?
Would a solution be that ExampleBrowser be available for newer platforms in homebrew, and disabled for older?

scpeters, 2015-05-06:
I would recommend 2.83.1 (or 2.83.0), while keeping the existing 2.83 tag. It's good to keep them for historical reference, even if they are broken.

Also, I'm using cmake with homebrew (see [lines 75-77 in the formula](https://github.com/scpeters/homebrew/blob/bullet_2_83/Library/Formula/bullet.rb#L75-L77)). The `--with-demo` option is a shortcut for enabling or disabling demos using the homebrew command line tool. I added some logic in the formula to disable the demos on 10.8 and earlier.

erwincoumans, 2015-05-06:
Can you delay the homebrew one day, I commit a fix that let's you run the
examples on older Mac OSX versions using OpenGL2. Testing that right now.

Thanks,
Erwin

scpeters, 2015-05-06:
Sure, I'll close the pull request for now.

erwincoumans, 2015-05-07:
ok, does the Bullet 2.83.1 release work for you?

closed 2015-05-07 by erwincoumans

scpeters, 2015-05-07:
Looks good so far, I've submitted it as https://github.com/Homebrew/homebrew/pull/39458

scpeters, 2015-05-07:
Actually I've got a compilation error on my yosemite machine when I do a plain cmake build (no options):
~~~
examples/OpenGLWindow/MacOpenGLWindow.mm:1039:5: error: use of
      undeclared identifier 'glSwapAPPLE'
    glSwapAPPLE();
    ^
1 error generated.
make[2]: *** [examples/OpenGLWindow/CMakeFiles/OpenGLWindow.dir/MacOpenGLWindow.o] Error 1
make[2]: *** Waiting for unfinished jobs....
~~~

scpeters, 2015-05-07:
It works if I build shared libraries but not static libraries

erwincoumans, 2015-05-07:
Aargh, cmake compiles just fine on 10.6.8, so they moved glSwapApple to gl3.h, but that header doesn't exist on my machine. What a waste of time, I'll see how to get around this.

erwincoumans, 2015-05-07:
Could you try adding #include <OpenGL/glext.h> in Bullet/examples/OpenGLWindow/OpenGLInclude.h in the Mac section, and see if that helps?
So it would look like:

#if defined (USE_OPENGL2) || defined (NO_OPENGL3)
#include <OpenGL/gl.h>
#include <OpenGL/glext.h>
#else
#include <OpenGL/gl3.h>
#endif

erwincoumans, 2015-05-07:
{% highlight c++ %}#include <iostream>{% endhighlight %}