bulletphysics/bullet3 #377: Setup CI for OS X

scpeters, 2015-05-07:
Bullet currently has CI for Ubuntu 12.04 (travis) and Windows (appveyor) but not OS X. Travis has beta support for OS X, but is [currently at capacity and not accepting new projects](http://docs.travis-ci.com/user/multi-os/).

Another option is to use jenkins. We have a mac mini as OSRF for CI that works quite well. Also, the homebrew project uses jenkins with 3 mac minis that run the three most recent OS releases (Yosemite, Mavericks, Mountain Lion).

There's no easy/free option, but Jenkins can work with appropriate hardware and maintenance.

erwincoumans, 2015-05-07:
Actually, the multi-OS feature was enabled for Bullet back in 2014, so I need to check it out if it still works. I'll see if adding - osx to the travis file works.

erwincoumans, 2015-05-07:
ok, first pull request is now building on Mac too, for travis, see
https://travis-ci.org/bulletphysics/bullet3/builds/61684139

erwincoumans, 2015-05-07:
Mac setup done using travis, seems to work

closed 2015-05-07 by erwincoumans