bulletphysics/bullet3 #389: MinGW segmentation fault when building

demystificator, 2015-05-19:
Hi,

I encounter a segmentation fault everytime at the same place 

<quote> C:\Bullet\git\src\BulletCollision\CollisionDispatch\btInternalEdgeUtility.cpp:310:2: internal compiler error: Segmentation fault</quote>

I'm building on :
Win7 64bits
GCC : 4.8.1 (MinGW32)
CMake : 3.1.0
Trying to build only the core lib.

Re-dl / reinstalling my compiler doesn't change anything. The same goes with bullet version (tried from zip and directly with git hub).

I found this : https://code.google.com/p/bullet/issues/detail?id=790&can=1&q=segmentation%20fault&colspec=Modified%20ID%20Type%20Stars%20Status%20Owner%20Summary

It is the closest topic to my problem I could find (and the most recent).

Hope it'll get fixed (or I'd be told why did I miss).

erwincoumans, 2015-05-19:
You are using a buggy compiler. If you have a workaround for your buggy compiler that makes Bullet compile, can you please share it? Otherwise, just delete the btInternalEdgeUtility.cpp and you should be good to go.

demystificator, 2015-05-28:
Well, sorry couldn't find any workaround so I just deleted the file :/ 
Hope it'll get fixed asap.

closed 2015-10-09 by erwincoumans