bulletphysics/bullet3 #394: Invalid cast when building with MinGW-w64

markand, 2015-05-27:
I'm trying to build bullet 2.83.4 on Windows using MinGW-w64 with GCC 4.9.

I get the following error while building:

````
[ 67%] Building CXX object examples/ExampleBrowser/CMakeFiles/App_ExampleBrowser.dir/__/__/Extras/Serialize/BulletFileLoader/bFile.obj
C:\src\64\bullet3-2.83.4\Extras\Serialize\BulletFileLoader\bFile.cpp: In member function 'void bParse::bFile::swapDNA(char*)':
C:\src\64\bullet3-2.83.4\Extras\Serialize\BulletFileLoader\bFile.cpp:446:13: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
   nr= (long)cp;
             ^
C:\src\64\bullet3-2.83.4\Extras\Serialize\BulletFileLoader\bFile.cpp:483:13: error: cast from 'char*' to 'long int' loses precision [-fpermissive]
   nr= (long)cp;
````

johnfredcee, 2016-02-03:
long is not big enough to hold a pointer : one possible fix is to change these to long long ; there are only a few sites where it is a problem.

johnfredcee, 2016-02-03:
[mingw64-fix.txt](https://github.com/bulletphysics/bullet3/files/116064/mingw64-fix.txt)

markand, 2016-02-04:
What about using (u)intrptr_t which is designed for?

johnfredcee, 2016-02-04:
No idea. Is C++11 specific code ok? The code base looks old and hairy, so I went for an old and hairy solution!

erwincoumans, 2016-02-04:
Sure, I'll look into it.

intptr_t has some portability issues but it only requires C99 not C++11, see also http://stackoverflow.com/questions/9492798/using-intptr-t-instead-of-void

The Bullet code base values portability and support for older compilers and platforms, so C++11 and C++14 is not allowed just yet. I rather have the code base look old and hairy rather than getting compile errors on various platforms.

johnfredcee, 2016-02-04:
That's *C* 99, not C++ - those were .cpp files I modified..

erwincoumans, 2016-02-04:
That's what I told indeed, C99. Some C++ implementations support C99, others don't. You don't need to have C++11 for that. Even in C++11 the intptr_t is optional, hence non-portable.

markand, 2016-02-05:
Le 04/02/2016 22:38, erwincoumans a écrit :
>
> The Bullet code base values portability and support for older 
> compilers and platforms, so C++11 and C++14 is not allowed just yet. I 
> rather have the code base look old and hairy rather than getting 
> compile errors on various platforms.
>

Deprecated platforms.

Also, a check can be quickly done with CMake to see if a specific 
typedef exists :-)

     include(CheckTypeSize)
     check_type_size(uintptr_t HAVE_UINTPTR_T)

Then you can easily add your own implementation if uintptr_t is missing.

-- 
David Demelier

closed 2019-04-29 by markand