bulletphysics/bullet3 #400: Namespace the library

aselle, 2015-06-03:
It would be extremely awesome if bullet was namespaced. Multiple shared libraries that use (and export) bullet libraries can conflict with each other. Yes, I know that I can use -fvisibility to make everything not export and use a static bullet library. However, that solution is extremely inconvenient for the other libraries I am using besides bullet. It seems like most OSS graphics software has settled on this convention. I know from a previous post there was hesitation to default this to on, and I think that is fine, but to have the infrastructure setup ready to go would be excellent. We'd be willing to help contribute a patch to this effect if there was some reasonable expectation of getting it into the mainline (because I am going to have to do it anyway)

erwincoumans, 2015-06-04:
Can you explain how the name spaces would avoid conflicts for multiple shared libraries with exported Bullet libraries? Would you rename the namespace for each shared library that uses of Bullet?

aselle, 2015-06-04:
Yes, you'd define a namespace for your personal bullet library. Then two copies of bullet could exist in the same process and dynamic symbol space.  For example, I've done this to bullet 2.7 where my build at Disney is called DisneyBullet so that it doesn't conflict with the non namespaced version that is being exposed in Houdini. Without that I get crashes.

erwincoumans, 2015-06-04:
I am happy to accept a contribution for an optional namespace, as long as namespaces are disabled by default and doesn't mess with anything (build systems, doxygen etc).
Also a patch should not make any other changes aside from adding the namespace macros (no change in layout, converting tabs to spaces, remove warnings).

AndresTraks, 2015-06-05:
Here are previous discussions about namespaces and code style in general:
http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?t=2723
http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?t=7600
http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?t=9642

On the topic of code style, some time ago there was an effort to clean up the formatting of the 2.xx code: https://code.google.com/p/bullet/issues/detail?id=110
There are still currently long lines, tabs in arbitrary places like method declarations and inconsistent indenting. Is there any chance that patches like these could be applied? http://pastebin.com/NJHfShZm

erwincoumans, 2015-06-05:
We need to review and apply outstanding patches in the issue tracker/pull request before making changes in the entire codebase, such as optional namespace, style improvements, inconsistencies etc. 

Bullet is mainly a 1-man development project (me) and occasional contributors (see push requests/issues). I can see how inconsistent indentation, tabs, spaces are a hassle for someone who wants to add functionality inside Bullet. If someone who makes an actual contribution that adds functionality to Bullet, I am more likely to accept style improvements.

Long lines don't bother me (I've been using wide-screen monitors for many years now, and in my opinion 80-line limits are prehistoric) but excessive long lines are not necessary indeed. If you think it is useful, we can pick some more relaxed maximum line width such as 120 characters, would that help?

By the way, I'm preparing another SIGGRAPH 2015 course about Bullet and likely create some slides with high-level information about which algorithms are used where.

aselle, 2016-06-20:
I have since left Disney, so I have no access to the patch anymore to cleanup and submit.

closed 2016-06-20 by aselle