bulletphysics/bullet3 #405: Intel HD graphics fails when running bulletphysics example browser

simo-11, 2015-06-10:
Hi

It seems that current texture allocation causes issues at least in lenovo T540p.
Change of  shadowMapWidth and shadowMapHeight from 8192 to 4096 in
GLInstancingRenderer.cpp makes application runnable.

Some details below.

Simo

D:\GitHub\bullet3>bin\App_ExampleBrowser_vs2010_debug.exe 
Version = 4.0.0 - Build 9.18.10.3272
Vendor = Intel
Renderer = Intel(R) HD Graphics 4600
b3Printf: Selected demo: Basic Example
Warning: b3Error[..\..\examples\OpenGLWindow\GLInstancingRenderer.cpp,1356]:

Warning: Assert ..\..\examples\OpenGLWindow\GLInstancingRenderer.cpp:1356 (glGetError() ==GL_NO_ERROR)

After this glTexImage2D call err gets value 0x505 
shadowMapWidth and Height seem to be 8192 (0x2000) which should be ok but still cause GL_OUT_OF_MEMORY errors. Changing to 4096 (0x1000) eliminates issue.
glTexImage2D(GL_TEXTURE_2D, 0,GL_DEPTH_COMPONENT16, shadowMapWidth, shadowMapHeight, 0,GL_DEPTH_COMPONENT, GL_FLOAT, 0);
err = glGetError();

erwincoumans, 2015-06-25:
Thanks for the report. Perhaps we can do some checking for maximum texture sizes in OpenGL? Otherwise a check based on Vendor could help.

tinfoilboy, 2015-07-06:
If I'm correct with OpenGL, you can `glGetInteger(GL_MAX_TEXTURE_SIZE, &size);`, and that will return the maximum supported texture size for the current GPU, you'd also cache that value because it wouldn't change, I could be wrong though.

simo-11, 2015-07-10:
In my machine glGetIntegerv(GL_MAX_TEXTURE_SIZE, &size); gives 0x2000 but after glTextImage2D glGetError will return 0x505 if size is 0x2000.
[image]

[image]

Solving this may not be worth the trouble but users that run into this issue can change the size of shadowMap

tinfoilboy, 2015-07-10:
It would be much easier just to have a field which has the shadow map size
by default to a very low value, and allow the user to change it. Unless for
some reason the user shouldn't change the shadow map size, I don't see a
problem with that.

simo-11, 2015-07-10:
Having low shadow map size will make shadows look quite bad.

That happens already with 4096 but I compensated it by making 
shadowMapWorldSize smaller
float shadowMapWorldSize=5; // was 100

This is with shadow map size=4096 and shadowMapWorldSize=5
[image]

This is with shadow map size=256 and shadowMapWorldSize=5
[image]

If this causes issues for others some solution should probably be done. 
I can help in testing if someone comes up with solution that Erwin accepts.

erwincoumans, 2015-07-10:
Is there a point in the application where it crashes or asserts? We could add some logic that next time, it runs with lower resolution shadow-map size. It seems mainly an issue with some Intel GPUs, because AMD and NVIDIA GPUs usually have plenty of memory. Lowering the shadowmap resolution+size for Intel GPUs seems more attractive. Another option is a small startup screen. I prefer not to lower the quality of everyone else, just because of limited capability GPUs. Note that you can also run with --opengl2 then it will use stencil buffer shadows.

executionunit, 2015-07-10:
sorry to butt in but I would suggest something like:

    size = max_desired_size
    texture = 0
    while(size > 0)
        generate texture
        if texture generated then break
        size = size >> 1
    if texture then
        yay
    else
        tell user to not use intel GPUs and disable feature

that's what I do in my game for these kind of global buffers allocated on the GPU

simo-11, 2015-07-10:
As noted in initial report it asserts and getting 0x505 means that gl instance cannot recover.
Solution suggested above seems attractive.
I could do some testing and then report results.

simo-11, 2015-07-10:
This seems to work and and does not reduce quality if full size can be handled.
I could not make pull request of this change only but it is available as
https://github.com/simo-11/bullet3/commit/575033d1cfdcfa81847817a7e5619acc6398e9b3

Spec says that minimum size is 64 so else above is not handled.

erwincoumans, 2015-07-12:
applied sim0-11 fix, thanks everyone for report and suggested fixes:
https://github.com/bulletphysics/bullet3/pull/421

closed 2015-07-12 by erwincoumans