bulletphysics/bullet3 #406: Compute reaction forces for joints (mobilizers) in btMultiBody

erwincoumans, 2015-06-10:
At the moment, the joint reaction forces is not available for btMultiBody joints. 

It would be good to compute them and make them available, similar to the btJointFeedback in btTypedConstraints.

We would need to cache some information during the forward dynamics computation (in btMultiBody::stepVelocitiesMultiDof) , as described nicely by Michael Sherman in the SimBody engine here: https://github.com/simbody/simbody/blob/master/Simbody/src/SimbodyMatterSubsystemRep.cpp#L5551

erwincoumans, 2015-06-19:
A first implementation is in my fork, but some more work needs to be done for 'mobilizer' motor and limits (explicitly computing the required forces, rather than updating the acceleration immediately).
https://github.com/erwincoumans/bullet3/commit/89edc40d61b7b9d36233094393ed07ff5470376f

kingchurch, 2015-06-19:
Will the constraint (motor/limits) induced forces be separate from the joint reaction force ? The former is the forces along the axis of mobilizers while the latter are the reaction forces perpendicular to the mobilizer axis.  Is the former the same as "appliedImpulse" on constraints ? Since appliedImpulse is already throttled by "maxImpulse" on motors/limits, the existing implementation may suffice in many cases. The latter is currently lacking to implement "break-able" mobilizer joints for example.

erwincoumans, 2015-06-19:
All constraints (contacts, mobilizer motors and mobilizer limits, other pair-wise constraints between btMultiBody and other btMultiBody or btRigidBody) will induce joint reaction forces. The way I implement it now is that all constraints will be added as 'external' forces to the btMultiBody. All those forces will be propagated in the articulated body algorithm and generate force/torque feedback for each mobilizer (btMultiBody internal joint). Does that work for you?

erwincoumans, 2015-06-19:
A new commit to support the motor/limits of the btMultiBody mobilizers (applying a force along the axis), and a pull request: https://github.com/bulletphysics/bullet3/pull/410
Still needs some testing, but mechanisms seems to work.

erwincoumans, 2015-06-25:
a first comparison between joint force/torque feedback using rigid body versus multi body looks promising, results are fairly similar. push request is here: https://github.com/bulletphysics/bullet3/pull/411
More testing and review is needed for the btMultiBody joint feedback

erwincoumans, 2015-07-14:
The feature has been implemented. See examples/MultiBody/TestJointTorqueSetup how to use it. Separately, there is also feedback for a multibody constraint, such as a hinge motor. See examples/MultiBody/MultiBodyConstraintFeedback.cpp

closed 2021-04-26 by erwincoumans