bulletphysics/bullet3 #407: bullet crashes at getOpenGLSubMatrix when using NEON

lefty2, 2015-06-11:
I found a "bad access" crash at the following line:
vm[0] = v0;
in btMatrix3x3::getOpenGLSubMatrix
it only happens when NEON is enabled.

erwincoumans, 2015-06-11:
Did you make sure your destination matrix is 16-byte aligned?

lefty2, 2015-06-11:
What? The distination matrix is 16 floats

erwincoumans, 2015-06-11:
ATTRIBUTE_ALIGNED16(float m[16]);

lefty2, 2015-06-11:
Isn't it already aligned without the macro?

erwincoumans, 2015-06-12:
Apparently it is not, so you need to align your data for this API call.
Indeed, it would be great if NEON would work on un-aligned data. I don't have the time create a work-around, so please use the alignment macro.

erwincoumans, 2015-07-12:
if alignment doesn't fix it, please provide a reproduction case and file a new issue. Thanks!

closed 2015-07-12 by erwincoumans