bulletphysics/bullet3 #409: btSimpleDynamicsWorld default gravity

jhemdan, 2015-06-18:
I'm not sure if this was intentional (there were no comments or anything explaining this that I've found) but the default gravity of btSimpleDynamicsWorld is set to btVector3(0, 0, -10) when logically it should be btVector3(0, -10, 0). I don't know if this was an oversight or not, but it caused me a period of confusion as to why my objects were falling in the forward direction.

Ehsanizadi, 2015-06-19:
To me it seems quite logical. As I always set Z axis as vertical axis in my dynamics world.
But its not a serious issue, you can always set it to a different value when you want to init your world.

AndresTraks, 2015-06-19:
Some graphics systems are Y-up, some are Z-up. In most cases it's easy to change, so use whichever feels natural. Just make sure the graphics subsystem, the physics subsystem and the model assets use the same setting.

Both make sense depending on where you come from. If you draw a graph on paper, the X-axis goes to the right and the Y-axis goes up. But if you draw a world map and try to map it to the real world, the X-axis goes across the land in one direction and the Y-axis goes across the land in another direction perpendicular to the X-axis. Now if you want to express elevation, all you're left with is the Z-axis, so you end up with Z pointing up.

jhemdan, 2015-06-19:
Yes, I'm aware that there are other coordinate systems that people use besides my own considering the fact that I had to deal with this when I convert models/skeletons from blender's Z-up coordinate system to my engine's. 

However, I've decided that this issue should be closed because I just checked the header file of btSimpleDynamicsWorld again and it states that it is a unit-test class and one should use btDiscreteDynamicsWorld instead (which uses the Y-axis by default). I guess when testing btSimpleDynamicsWorld, the author decided to test the gravity on a different axis.

Anyway, I blame whatever tutorial I followed when I first used Bullet a couple years ago for telling me to use btSimpleDynamicsWorld even though the documentation states not to.

closed 2015-06-19 by jhemdan