bulletphysics/bullet3 #416: Accuracy of Bullet: no of itteration vs time step?

Ehsanizadi, 2015-07-06:
Hi,

I'm working on simulation of particulate media when subjected to different stresses (normal and shear) using Bullet. Within the context of my work I have to also measure the accuracy of Bullet.

My simulation involves about 2000 densely positioned spheres subjected to shear and normal stresses. I tried to play with two parameters to see the effects of them on the results. 
### What is my result?
I mainly focused on contact forces and I plot such forces in form of stress and displacement. So, the x axis would be shear displacement, which is a kind of input to our problem, and y-axis is the shear stress, which is the output of the simulation. The nature of the response is in such a way that at very low shear displacements, there is a high shear stress, but with further increase in shear displacement, the whole system fails and the shear stress tends to decrease as shear displacement increases.

#### The effect of number of iteration
The effect of number of iterations on the shear stress is demonstrated in the figure below:
The time step for all cases is taken to 0.001 seconds.

[image]
 As it can bee seen when the number of iterations is taken to 20, which is the bullet default value, there are many fluctuations in the graph, however, with increase in the number of iterations, the bound of fluctuations tend to decrease (which is good) but with further increase in number of iteration, this bound will not be more narrower!

#### The effect of time step
The effect of time step on the results also is investigated, as shown below.
[image]

Well, for delta-t = 10x,  the results are quite shaky but with decrease in delta-t, the results will be so nice and with minimized fluctuation bound. That's good! 
But the strange thing is that why with more decreasing of delta-time (delta-t/2 and delta-t/10) the fluctuations appear again and the results become more inaccurate?

----------------------------------
Could it be due to the fact that force/stress is not a correct parameter to measure the accuracy?
For example the forces can be derived from position of the objects by putting spring between the objects I already measured the forces for them. Below is the same graph as two previous ones, but for this one I derived the forces from spring, which is position dependent:
[image]
 
Note that negligible change observed when changing the iteration number or delta time of the simulation in the latter case.

If the answer is yes (i.e. force is not the correct parameter to measure the accuracy since they are computed externally, or due to integration or etc.) how I can measure the accuracy of the Bullet within the force/stress framework?

Thanks for your help in advance.

erwincoumans, 2015-07-06:
Can you implement tests in C++ and create a pull request? Make sure to use double precision when using a small timestep. For general discussion please use the Bullet forums. Thanks!

Ehsanizadi, 2015-07-09:
You can find the tests in C++ here: https://github.com/Ehsanizadi/Bullet-examples/tree/master/directShearTest

With double precision turned on, very negligible changes happen to plots. So, the issue still exists.

Also another post regarding reducing of timestep reported in Bullet forums : http://www.bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=10695&p=35899#p35899

erwincoumans, 2015-07-14:
The code in your repository won't compile because it has different folder layout and unix specific dependencies. If you can modify an origin Bullet example (in Bullet/examples directory) without adding platform/compiler specific dependencies, I will have a look at it.

Ehsanizadi, 2015-07-28:
To simplify everything, forget about my code (since it is complex and composed of thousands objects), 
you can open the famous helloWorld example and set the time increment to 0.01, and restitution of both falling and static object to 1.  Let the fall object to free fall from a certain height. monitor the position of the fall object. it'll be nice, but if you set the time increment to 0.001, or less, the fall  object wont bounce any more!

It is reported also here: http://www.bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=10695&p=35903#p35903

erwincoumans, 2015-07-31:
The effect you describe in the HelloWorld with varying timestep is likely due to continuous collision detection. I'll look at it after the SIGGRAPH conference.

erwincoumans, 2015-09-11:
Ping, I need to look at this. Still too busy with other things.

closed 2021-04-26 by erwincoumans