bulletphysics/bullet3 #423: Problem making HelloWorld.cpp

AjayTalati, 2015-07-12:
Hi, I'm trying to compile the `HelloWorld.cpp`

When I run `make`, after `cmake .` I get the error,

```unix
[100%] Building CXX object CMakeFiles/App_HelloWorld.dir/HelloWorld.o
/home/ajay/BulletPhyProjects/bullet3/examples/HelloWorld/HelloWorld.cpp:17:36: fatal error: btBulletDynamicsCommon.h: No such file or directory
 #include "btBulletDynamicsCommon.h"
                                    ^
compilation terminated.
make[2]: *** [CMakeFiles/App_HelloWorld.dir/HelloWorld.o] Error 1
make[1]: *** [CMakeFiles/App_HelloWorld.dir/all] Error 2
make: *** [all] Error 2
```

My my [CMakeList](https://github.com/bulletphysics/bullet3/blob/master/examples/HelloWorld/CMakeLists.txt) is the same as in the repo. 

I've got the [basic build](https://github.com/bulletphysics/bullet3#build-instructions-for-bullet-3) working for Linux 64, and also ran `cmake` and `make` in the root directory, so that 

```
./bullet3/src/Bullet3Dynamics/libBullet3Dynamics.a 
./bullet3/src/Bullet3Collision/libBullet3Collision.a 
./bullet3/src/LinearMath/libLinearMath.a 
```

Have all been built. Not sure what I've missed?

AjayTalati, 2015-07-12:
I'm sure I understand this, but I'm making to compile the program with,

`g++ ./bullet3/examples/HelloWorld/HelloWorld.cpp -lGL -lGLU -I ./bullet3/src ./bullet3/src/BulletDynamics/libBulletDynamics.a ./bullet3/src/BulletCollision/libBulletCollision.a ./bullet3/src/LinearMath/libLinearMath.a`

opening the output,

`./a.out`

gives the numbers? I don't understand why I get the error when I try to make the project files, but it still compiles as expected - it's really confusing ???

erwincoumans, 2015-07-12:
please do not use the issue tracker for discussions like this. you can use the Bulet forums at http://bulletphysics.org

closed 2015-07-12 by erwincoumans