bulletphysics/bullet3 #447: Official Readme: Please add example how to enable .so without .a via cmake

shevegen, 2015-08-05:
Hi,

I am compiling bullet for enlightenment efl.

Unfortunately efl fails due to a bullet*.a file, where I don't have -fPIC enabled.

In GNU autoconfigure I can use --disable-static --enabled-shared then I won't have .a files, but I have no idea how to do this via cmake.

Could this be added to the official README or INSTALL file so that people can understand how to get this via cmake as well? Thanks!

scpeters, 2015-08-05:
~~~
mkdir build
cd build
cmake .. -DBUILD_SHARED_LIBS=ON
~~~

See also the [bullet homebrew formula](https://github.com/Homebrew/homebrew/blob/master/Library/Formula/bullet.rb#L37).

scpeters, 2015-08-05:
`BUILD_SHARED_LIBS` is a standard cmake variable:

* http://www.cmake.org/cmake/help/v3.0/variable/BUILD_SHARED_LIBS.html

scpeters, 2015-08-05:
It could still go in the readme

erwincoumans, 2019-12-18:
Closing out-of-date issues. BUILD_SHARED_LIBS  is standard cmake.

closed 2019-12-18 by erwincoumans