bulletphysics/bullet3 #450: Fix the Bullet Raycast sample.

benelot, 2015-08-06:
The first hit example is missing a

`m_dynamicsWorld->rayTest(from,to,closestResults);`

so instead of

			btVector3 from(-30,1.2,0);
			btVector3 to(30,1.2,0);
			m_dynamicsWorld->getDebugDrawer()->drawLine(from,to,btVector4(0,0,1,1));

			btCollisionWorld::ClosestRayResultCallback	closestResults(from,to);
			closestResults.m_flags |= btTriangleRaycastCallback::kF_FilterBackfaces;
			
			if (closestResults.hasHit())
			{
				
				btVector3 p = from.lerp(to,closestResults.m_closestHitFraction);
				m_dynamicsWorld->getDebugDrawer()->drawSphere(p,0.1,blue);
				m_dynamicsWorld->getDebugDrawer()->drawLine(p,p+closestResults.m_hitNormalWorld,blue);

			}

it should be

			btVector3 from(-30,1.2,0);
			btVector3 to(30,1.2,0);
			m_dynamicsWorld->getDebugDrawer()->drawLine(from,to,btVector4(0,0,1,1));

			btCollisionWorld::ClosestRayResultCallback	closestResults(from,to);
			closestResults.m_flags |= btTriangleRaycastCallback::kF_FilterBackfaces;
			
			m_dynamicsWorld->rayTest(from,to,closestResults);

			if (closestResults.hasHit())
			{
				
				btVector3 p = from.lerp(to,closestResults.m_closestHitFraction);
				m_dynamicsWorld->getDebugDrawer()->drawSphere(p,0.1,blue);
				m_dynamicsWorld->getDebugDrawer()->drawLine(p,p+closestResults.m_hitNormalWorld,blue);

			}

benelot, 2015-08-07:
Ok I saw that it was an issue with an old revision I had, it seems to be there already.

benelot, 2015-08-07:
It is missing here:
https://github.com/bulletphysics/bullet3/blob/master/examples/Raycast/RaytestDemo.cpp

erwincoumans, 2015-08-10:
thanks for the feedback, we'll fix it

benelot, 2015-08-25:
I made you a pullrequest:
https://github.com/bulletphysics/bullet3/pull/464

PR #464 (2015-10-09): Fix the Bullet Raycast sample. Closes #450.
This closes #450.

The first hit example is missing a

`m_dynamicsWorld->rayTest(from,to,closestResults);`

so instead of

			btVector3 from(-30,1.2,0);
			btVector3 to(30,1.2,0);
			m_dynamicsWorld->getDebugDrawer()->drawLine(from,to,btVector4(0,0,1,1));

			btCollisionWorld::ClosestRayResultCallback	closestResults(from,to);
			closestResults.m_flags |= btTriangleRaycastCallback::kF_FilterBackfaces;
			
			if (closestResults.hasHit())
			{
				
				btVector3 p = from.lerp(to,closestResults.m_closestHitFraction);
				m_dynamicsWorld->getDebugDrawer() [... cut]
commit d1d279d0 (2015-10-09): Merge pull request #464 from benelot/master

Fix the Bullet Raycast sample. Closes #450.
commit e39bf6c4 (2015-10-09): Fix the Bullet Raycast sample. Closes #450.
closed 2015-10-09 by erwincoumans