bulletphysics/bullet3 #459: btCylinderShape rayTest gives hit when ray starts from inside

lefty2, 2015-08-18:
If you do a ray test with the ray starting inside a btCylinderShape most of the time the ray does not hit the cylinder (seemingly it's back culled), but every now and again (maybe 1 in 1000 times) it does hit. That seems to be a bug for me - it should either hit all the time, or never.

benelot, 2015-08-18:
I tried the same with other shapes, in my case the box and the capsule shape and I also do not get any hits from inside. 

I even tried to remove the filter backfaces in case that was the problem, but it does not change it at all.

     rayCallback.m_flags &= !btTriangleRaycastCallback::kF_FilterBackfaces;

My solution here was just to invert the ray to get proper hits. But it should be looked into from the Bullet Physics side.

lefty2, 2015-08-18:
The fact the ray does not hit from inside is not the problem. The problem is that it sometimes hits and sometimes doesn't on a seemly random basis.
Also, inverting the ray is not a solution, unless you are using AllHitsRayResultCallback.
It's not clear if they ment to use back culling or not. The kF_FilterBackfaces is not used at all in the source code.

erwincoumans, 2015-09-11:
Yes, this should be addressed. We lack resources to address this at the moment and fix it once we have time.

benelot, 2015-11-20:
I changed the raytest example to show the issue:
http://bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=10924

closed 2021-04-26 by erwincoumans