bulletphysics/bullet3 #468: Fix or disable specific warnings on various platforms

erwincoumans, 2015-09-02:
The Bullet code base generates many warnings on various compiles (and platforms). We should fix them or suppress them in cmake and premake.

tommimarkus, 2015-12-07:
With these settings:

		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_WARNING_CFLAGS "-Wreorder")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_WARNING_LDFLAGS " ")
		# C/C++
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_CXX0X_EXTENSIONS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_INVALID_OFFSETOF_MACRO "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_HIDDEN_VIRTUAL_FUNCTIONS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_NON_VIRTUAL_DESTRUCTOR "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN__EXIT_TIME_DESTRUCTORS "NO")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNUSED_VARIABLE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNUSED_VALUE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNUSED_PARAMETER "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNUSED_LABEL "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNUSED_FUNCTION "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_UNREACHABLE_CODE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNKNOWN_PRAGMAS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_UNINITIALIZED_AUTOS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_TYPECHECK_CALLS_TO_PRINTF "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_TREAT_IMPLICIT_FUNCTION_DECLARATIONS_AS_ERRORS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_TREAT_INCOMPATIBLE_POINTER_TYPE_WARNINGS_AS_ERRORS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_SUSPICIOUS_IMPLICIT_CONVERSION "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_SIGN_COMPARE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_POINTER_SIGNEDNESS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_ASSIGN_ENUM "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_MISSING_NEWLINE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_MISSING_PROTOTYPES "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_MISSING_FIELD_INITIALIZERS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_MISSING_PARENTHESES "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_RETURN_TYPE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_INITIALIZER_NOT_FULLY_BRACKETED "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_IMPLICIT_SIGN_CONVERSION "NO")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_NULLABLE_TO_NONNULL_CONVERSION "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_INT_CONVERSION "$(CLANG_WARN_SUSPICIOUS_IMPLICIT_CONVERSION)")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_ENUM_CONVERSION "$(CLANG_WARN_SUSPICIOUS_IMPLICIT_CONVERSION)")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_CONSTANT_CONVERSION "$(CLANG_WARN_SUSPICIOUS_IMPLICIT_CONVERSION)")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_BOOL_CONVERSION "$(CLANG_WARN_SUSPICIOUS_IMPLICIT_CONVERSION)")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_64_TO_32_BIT_CONVERSION "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_SHADOW "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_FOUR_CHARACTER_CONSTANTS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_EMPTY_BODY "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_DOCUMENTATION_COMMENTS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ABOUT_DEPRECATED_FUNCTIONS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_CHECK_SWITCH_STATEMENTS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_TREAT_WARNINGS_AS_ERRORS "NO")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_PEDANTIC "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_INHIBIT_ALL_WARNINGS "NO")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_SHORT_ENUMS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_ENABLE_BUILTIN_FUNCTIONS "YES")
		# Obj-C
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_DIRECT_OBJC_ISA_USAGE "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN__DUPLICATE_METHOD_MATCH "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_OBJC_IMPLICIT_ATOMIC_PROPERTIES "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_OBJC_MISSING_PROPERTY_SYNTHESIS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_ALLOW_INCOMPLETE_PROTOCOL "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_MULTIPLE_DEFINITION_TYPES_FOR_SELECTOR "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_DEPRECATED_OBJC_IMPLEMENTATIONS "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_STRICT_SELECTOR_MATCH "YES")
[... cut ...]
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_OBJC_IMPLICIT_RETAIN_SELF "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_OBJC_REPEATED_USE_OF_WEAK "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN_OBJC_RECEIVER_WEAK "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_CLANG_WARN__ARC_BRIDGE_CAST_NONARC "YES")
		set_property (TARGET ${target} PROPERTY XCODE_ATTRIBUTE_GCC_WARN_CHECK_SWITCH_STATEMENTS "YES")

I get 6340 warnings from bullet3 :-/

erwincoumans, 2015-12-08:
Thanks tommimarkus, do you have a patch/pull request for this? Where do you put those lines in the CMakeLists.txt files? Is set_property(TARGET the same as SET_TARGET_PROPERTY(... ?

tommimarkus, 2015-12-11:
As far as I know `set_property(TARGET...` is the same as `set_target_property(...`
This is a function that I call and it's pretty much the last thing on my cmake configs. (CMakelists.txt includes a bunch of .cmake files and it is in last of those)

erwincoumans, 2015-12-11:
Do you have a patch that applies this to the existing Bullet CMakeLists.txt for all projects/libraries?

josiahls, 2019-04-05:
Is there a non-cmake level way to suppressing warnings? Such as a default boolean on the method calls? Or are we not doing this due to performance reasons?

``` removeBody(index, true) ```

I get warnings when removing bodies that dont exist, and I seem to have to do it this way because I have objects being added and removed thus cluttering the "numBodies". Calling removeBodies on a range of indexes seems to work best for me, but I get a lot of warnings out of it.

erwincoumans, 2019-04-05:
@josiahls Your warning are run-time warnings in PyBullet. The other are C++ compile warnings, different. We could add some method to suppress warnings in PyBullet.

closed 2021-04-26 by erwincoumans