bulletphysics/bullet3 #482: Enabling SSE potentially slows down performance(?)

juj, 2015-10-07:
I'm experimenting with the Bullet physics engine version 2.82, and comparing scalar code implementation to SSE implementation (`BT_USE_SSE` and `BT_USE_SSE_IN_API` defined). My tests are being run on a Windows Haswell system with Visual Studio 2015 compiler. The scene consists of 1500 boxes falling from the sky in six tall columns. You can find an interactive HTML5 web build of the scene here: https://dl.dropboxusercontent.com/u/40949268/emcc/2015-10-05-emunittest_0.4-PhysicsStressTest-Urho3D_1.4_e1.34.11-release-prof/12_PhysicsStressTest.html (please run in Firefox, preferably in 64-bit and in Nightly for best performance). Note that the web build is just for illustration purposes of what my scene looks like. I am not benchmarking the web build, but a native Windows 64-bit build.

The results I am getting are quite surprising: when I compile a 32-bit Windows executable with `BT_USE_SSE` and `BT_USE_SSE_IN_API` defined, I am getting roughly 10% *worse* performance compared to the scalar build. When comparing 64-bit builds, SSE is a bit faster than scalar, but only about 3% (tested one run, not much for statistical significance). I am profiling the hotspots with AMD CodeXL, and in each case, Bullet functions are consuming more than 90% of the total frametime, so the 3D engine driving the application should not be affecting the results. Looking at AMD CodeXL profiles confirms that the workload is saturated with Bullet simulation:

   - scalar: https://dl.dropboxusercontent.com/u/40949268/dump/Urho3D_scalar.png
   - sse: https://dl.dropboxusercontent.com/u/40949268/dump/Urho3D_sse.png

In the web build, that can be confirmed by hitting F2, and observing the times given by the internal profiler, "StepSimulation" is where Bullet physics is being driven.

I have fixed the timesteps I pass to `stepSimulation`, calling it at a constant rate of timeStep = 0.016667, maxSubSteps=0, internalTimeStep = 0.016667.

My question is whether there exists benchmarks of Bullet's sse performance posted somewhere? Are my results to be expected?

juj, 2015-10-07:
Ops, very sorry, the flags I had been testing with in above were `BT_USE_SSE`+`BT_USE_SIMD_VECTOR3`, and not `BT_USE_SSE`+`BT_USE_SSE_IN_API`. I tried rerunning the tests with `BT_USE_SSE`+`BT_USE_SIMD_VECTOR3`+`BT_USE_SSE_IN_API`, but the results were practically the same.

erwincoumans, 2015-10-07:
I would recommend using the latest Bullet (which is 2.83.5), why would you use an old version 2.82?

My guess: the SSE/SIMD implementation was written many years ago, when compilers didn't do auto vectorization. Nowadays, compilers perform auto vectorization and automatically optimize for SSE/SIMD.

Nowadays, it is better to optimize for memory access instead of just clock cycles.

juj, 2015-10-07:
Thanks, I was using 2.82 since the engine I'm working with is currently integrated with that version. I didn't realize that 2.82 was actually so old, from October 2013. I'll see about upgrading to see how it behaves.

juj, 2015-10-08:
I upgraded to Bullet 2.83.6, and reran the tests, but the results were much the same, in SSE2, I get a time of 25.648 seconds (averaged over three runs of the test), and in scalar, I get 25.207 seconds (averaged over three runs). Based on your comment, it does sound expected that SIMD speedup does not apply to modern CPUs/current compilers? Or do you think it would be more a case of the current SIMD codebase just being "out of tune" on VS2015/Haswell?

erwincoumans, 2015-10-08:
My guess is that latest modern compilers automatically create SIMD code from regular C,C++ code, so manual SIMD as used in a few parts in Bullet is likely obsolete. More important, data optimizations are likely more important nowadays. Right now I don't have time for optimizations, so unless you have a patch ready, lets close this report.

closed 2015-10-08 by erwincoumans

juj, 2015-10-08:
No worries, thanks for the feedback!

erwincoumans, 2015-10-08:
Thanks for your report too. Right now I am working on some other areas of Bullet. At some stage I'll optimize the implementation...

tamaynard, 2015-10-13:
You mention AMD CodeXL. I believe running SSE on AMD was slower at some point than Intel, not sure if this is still the case or if it would explain the discrepancies you see here, but could be worth checking out if you haven't already done so.

velkyel, 2015-11-13:
> My guess is that latest modern compilers automatically create SIMD code from regular C,C++ code, so manual SIMD as used in a few parts in Bullet is likely obsolete. More important, data optimizations are likely more important nowadays.

Data optimizations are important of course, but..  On neon arch I have better results with NEON disabled in btVector3 math.  For example btGenerateInternalEdgeInfo is approx 20% faster.  I keep NEON vectorisation in minDot/maxDot only - there it make sense - it's batch operation and most of time is faster than scalar version.  Common btVector3 math like operator+ and others leads in real-app to frequent load/store float32x4_t which is slow.  Implementing SIMD in general vector/matrix math is not good idea.