bulletphysics/bullet3 #510: Friction in a btSlider / btHingeConstaint

kzahedi, 2015-10-29:
Is there way to add friction to a joint?

scpeters, 2015-10-29:
The approach we have taken in gazebo (see [issue 381](https://bitbucket.org/osrf/gazebo/issues/381/ode-joints-dont-support-friction-parameter#comment-12148418)) is to use the joint motor functionality. I set the desired velocity to zero and the motor force/torque to the friction value.

For the slider joint: https://bitbucket.org/osrf/gazebo/src/ba680b40dcadefb8bebb9835ae26b8b4d2c6cc0e/gazebo/physics/bullet/BulletSliderJoint.cc?fileviewer=file-view-default#BulletSliderJoint.cc-376

For the hinge joint: https://bitbucket.org/osrf/gazebo/src/ba680b40dcadefb8bebb9835ae26b8b4d2c6cc0e/gazebo/physics/bullet/BulletHingeJoint.cc?at=default&fileviewer=file-view-default#BulletHingeJoint.cc-398

Look at the code, because some functions take an impulse instead of a force/torque.

kzahedi, 2015-10-30:
Thank you very much for the quick reply. I will definitely have a look at the code in the next couple of days. I didn't really understand the difference of impulse force/torque in bullet. I couldn't find any documentation on it.

scpeters, 2015-10-30:
Well, the `btHingeConstraint` function is called [setMaxMotorImpulse](https://github.com/bulletphysics/bullet3/blob/5ded2d3ee437c74caccc136bf8c35bebf3e2fc62/src/BulletDynamics/ConstraintSolver/btHingeConstraint.h#L179), but the `btSliderConstraint` function is called [setMaxMotorImpulse](https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/ConstraintSolver/btSliderConstraint.h#L239). You can convert from force to impulse with a fixed-step solver by multiplying by the time-step size (see [impulse definition here](https://en.wikipedia.org/wiki/Impulse_%28physics%29)).

kzahedi, 2015-11-02:
Thank you very much for the link to the impulse wikipage. Now I know why I had to use ridiculously high values in my simulator :)

Thank you also very much for the link to the source code. That's a nice trick that I can easily adapt to my simulator.

scpeters, 2015-11-03:
I'm happy to help. Ready to close this issue?

erwincoumans, 2015-11-06:
Yes, a joint motor with target velocity to zero is a good approach for now. In a future revision of Bullet we may add built-in support for joint friction.

closed 2015-11-06 by erwincoumans