bulletphysics/bullet3 #520: Crash in  gjkepa2_impl::GJK::EncloseOrigin()

kingchurch, 2015-11-08:
Using HEAD of master as of Nov 7. Make a OSX 64-bit release build.  Run the "Rolling Friction"  demo and the Example Broswer crashed when the blocks hit the ground. See the backgract in the attached screenshot.

[image]

erwincoumans, 2015-11-08:
Thanks for the report. I can't reproduce such crash, on Mac OSX, 64bit, release build with Rolling Friction demo. What build system did you use, premake or cmake, all details/flags etc, please.
What version of the compiler (Xcode and clang version) and operating system version?

erwincoumans, 2015-11-08:
I could reproduce it, it might be some clang issue related to namespaces.
Can you see if this pull request fixes the issue for you too?
https://github.com/bulletphysics/bullet3/pull/521

erwincoumans, 2015-11-09:
Clang got confused by duplicate class names in Bullet3OpenGL and BulletCollision. Fixed in latest revision, see https://github.com/bulletphysics/bullet3/pull/521

closed 2015-11-09 by erwincoumans