bulletphysics/bullet3 #539: Why URDFRevoluteJoint has no btMultiBodyJointLimitConstraint in createMultiBody mode in URDF2Bullet.cpp

mirsking, 2015-11-29:
Hi,
I'm now working on grasp physics simulation using the kinova kg3 gripper.(This gripper has 6 dof, 2 in each finger.)
I  import the gripper's urdf into bullet ExampleBrower by URDF2Bullet.cpp.
Howerer, the simulation is just doesn't make sense.
I got things like this.
[image]

Then I find in URDF2Bullet.cpp file, URDFRevoluteJoint just doesn't have btMultiBodyJointLimitConstraint, which can have lower and upper limit of joint value.

When I add the constraint in the same way as that in URDFPrismaticJoint, the problem doesn't happy. I don't whether this is a bug.

However, when I add torque into each finger joint by 
```m_multiBody->addJointTorque(joint, force);``` in stepSimulation.
The joint only move a little, while there still are torque add to every joint.
The scene is like
[image]
And the torque is still not zero like this
```
current 0 pos: -0.053124
add 0 joint torque: 85.312431
current 1 pos: 0.416667
add 1 joint torque: 38.333332
current 2 pos: -0.000000
add 2 joint torque: 80.000015
current 3 pos: -0.066577
add 3 joint torque: 86.657715
current 4 pos: -0.053124
add 4 joint torque: 85.312431
current 5 pos: 0.416667
add 5 joint torque: 38.333332
```

Thank you for your help.

erwincoumans, 2015-11-29:
Can you fork the latest bullet3 repo and modify/add a demo with your grasp sim + URDF file,
and provide a link? Please make it build out-of-the-box similar to the existing Bullet demos, using either premake or cmake. Thanks!

mirsking, 2015-11-30:
Hi, I've push my codes here https://github.com/mirsking/bullet3/tree/graspit_demo  and there are two commits. The first one is without btMultiBodyJointLimitConstraint in URDFRevoluteJoint, and I've modify URDF2Bullet.cpp to add btMultiBodyJointLimitConstraint for URDFRevoluteJoint in the other commit.
Thank you for your help!

erwincoumans, 2015-12-04:
I had a look at your project, and there are many issues.
1) First of all, the collision meshes are way too detailed, and should be simplified a lot.
2) You should disable self-collision for the btMultiBody
3) It is best to not use any constraint limits. Just use 'continuous' joints without limits
4) The masses and inertias are extremely small and this will lead to instability
5) It is best to use joint motors to drive the fingers, as a first test. Once that works nicely, then consider using other control methods (computed torque + inverse dynamics, or PID control etc)

What is the license of the grasp example and the URDF file? Can it be added to Bullet?
I consider creating a working grasp demo, and this seems a nice starting point.

mirsking, 2015-12-05:
Thank you for your help. And I'll try with these instructions.
I have not decided the license right now. But it can be used freely and in any way you like, and also can be added into Bullet.

benelot, 2016-05-03:
Mirsking: Does your gripper work now? In case it is working, we could make an example from it and make apull request to include it into the Bullet Example Browser. Just tell us when you are ready.

mirsking, 2016-05-04:
Sorry, it's not ready yet. since I'm now using baxter gripper for the simulation, I'm not focusing on this gripper. But I think I will try this gripper later,  and when it's ok, I will reply here. Thanks for the attentation.

closed 2021-04-26 by erwincoumans