bulletphysics/bullet3 #540: [BulletInverseDynamicsUtils] Error 2

Schizo, 2015-12-02:
I tried to build it for my linux64 machine, but when running the make command i am getting following error:

```
Creating obj/x64/Release/BulletInverseDynamicsUtils
CoilCreator.cpp
In file included from ../../src/BulletInverseDynamics/IDConfig.hpp:65,
                 from ../../Extras/InverseDynamics/MultiBodyTreeCreator.hpp:6,
                 from ../../Extras/InverseDynamics/CoilCreator.hpp:4,
                 from ../../Extras/InverseDynamics/CoilCreator.cpp:3:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp: In constructor ‘btInverseDynamicsBullet3::vecx::vecx(int)’:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:51: error: class ‘btInverseDynamicsBullet3::vecx’ does not have any field named ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp: In member function ‘const btInverseDynamicsBullet3::vecx& btInverseDynamicsBullet3::vecx::operator=(const btVectorX<float>&)’:
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected type-specifier before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘>’ before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘(’ before ‘btVectorX’
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: missing template arguments before ‘*’ token
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected primary-expression before ‘>’ token
../../src/BulletInverseDynamics/details/IDLinearMathInterface.hpp:53: error: expected ‘)’ before ‘;’ token
make[1]: *** [obj/x64/Release/BulletInverseDynamicsUtils/CoilCreator.o] Error 1
make: *** [BulletInverseDynamicsUtils] Error 2
```

erwincoumans, 2015-12-02:
what compiler+version do you use? Did you use premake or cmake to create Makefiles? What linux distro?

(It works fine here on Linux 64 by the way)

erwincoumans, 2015-12-05:
Could it be you enabled EIGEN library instead of Bullet LinearMath?

cstoltz, 2016-08-02:
I'm having the same problem.  My versions are below:

Red Hat Enterprise Linux 6.6
Cmake – 3.6.1
Make – GNU Make 3.81
G++ - gcc version 4.4.7 20120313 (Red Hat 4.4.7-11)

Could anyone offer any help?

Thanks

cstoltz, 2016-08-02:
Note that we were able to compile on a workstation with the following:
Cmake - 3.4.3
Make - GNU Make 4.0
G++ - gcc version 5.3.1

Guessing that I likely need a more recent version of make and/or g++ on the first machine???

erwincoumans, 2016-08-02:
You likely are using some buggy gcc compiler version, older and newer gcc versions probably work.

erwincoumans, 2016-08-02:
Can you test if attached patch works around your broken compiler issue
[vecdiff.zip](https://github.com/bulletphysics/bullet3/files/397605/vecdiff.zip)

?

cstoltz, 2016-08-02:
I had to add a correction to one line 

inline mat33 operator/(const mat33& a, const idScalar& s) { return a * (1.0 / s); }

as the semicolon and curly bracket were missing at the end.  It does appear to have compiled.  Thanks!

erwincoumans, 2016-08-02:
can you send a patch for that?

cstoltz, 2016-08-02:
All I did was to add the few characters to the original patch you provided.

I made the changes to the bullet file manually rather than through an automated process.

[vecdiff.zip](https://github.com/bulletphysics/bullet3/files/397916/vecdiff.zip)

erwincoumans, 2016-08-04:
ok, patch was applied to github, let's close issue.

closed 2016-08-04 by erwincoumans

chengxuz, 2017-01-28:
Sorry, I still got this problem, and I think this is due to that the patch is somehow not there for current master branch. And after actually fixing the codes according to the patch (as things changed, the patch can not be used directly), I could solve this problem. So one need to re-patch this thing. The new patch I used is as below:

```
diff --git a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
index 5bb4a33..0d5ea3e 100644
--- a/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
+++ b/src/BulletInverseDynamics/details/IDLinearMathInterface.hpp
@@ -15,6 +15,7 @@ class vec3;
 class vecx;
 class mat33;
 typedef btMatrixX<idScalar> matxx;
+typedef btVectorX<idScalar> vecxx;

 class vec3 : public btVector3 {
 public:
@@ -47,11 +48,11 @@ inline mat33 operator/(const mat33& a, const idScalar& s) { return a * (1.0 / s)

 inline mat33 operator*(const idScalar& s, const mat33& a) { return a * s; }

-class vecx : public btVectorX<idScalar> {
+class vecx : public vecxx {
 public:
-	vecx(int size) : btVectorX(size) {}
-	const vecx& operator=(const btVectorX<idScalar>& rhs) {
-		*static_cast<btVectorX*>(this) = rhs;
+	vecx(int size) : vecxx(size) {}
+	const vecx& operator=(const vecxx& rhs) {
+		*static_cast<vecxx*>(this) = rhs;
 		return *this;
 	}

```