bulletphysics/bullet3 #541: Inverse dynamics not working with a floating base

DivideByZero2, 2015-12-02:
Created using btMultiBodyTreeCreator. Fixed base works correctly

erwincoumans, 2015-12-02:
The first inverse dynamics example code didn't support floating base.
I modified the example to support floating base, by adding 6 dummy values that represent the floating base. See https://github.com/bulletphysics/bullet3/pull/542

DivideByZero2, 2015-12-03:
Thanks for the example. It looks like I just had a bug in my code (forgot the extra 6 dummy values). I think there might be a small typo in the example. 

qdot6[6+i] = qdot6[i];
q6[6+i] = q6[i];
joint_force6[6+i] = joint_force6[i];
=>
qdot6[6 + i] = qdot[i];
q6[6 + i] = q[i];
joint_force6[6 + i] = joint_force[i];

closed 2015-12-03 by DivideByZero2