bulletphysics/bullet3 #552: bug in btMatrix3x3 equal operator

yyzone, 2015-12-18:
in SIMD_FORCE_INLINE bool operator==(const btMatrix3x3& m1, const btMatrix3x3& m2)

current code for SSE is:

    __m128 c0, c1, c2;

    c0 = _mm_cmpeq_ps(m1[0].mVec128, m2[0].mVec128);
    c1 = _mm_cmpeq_ps(m1[1].mVec128, m2[1].mVec128);
    c2 = _mm_cmpeq_ps(m1[2].mVec128, m2[2].mVec128);
    
    c0 = _mm_and_ps(c0, c1);
    c0 = _mm_and_ps(c0, c2);

    return (0x7 == _mm_movemask_ps((__m128)c0));

the last line should be changed to:

    int m = _mm_movemask_ps((__m128)c0);
    return (0x7 == (m & 0x7));

or am i wrong?

Joshhua5, 2016-06-03:
I don't work on this repository, but the version of the code you suggested would AND m and 0x7, the result from that would either be 0 if false or 0x7 if true, you can return the value here as anything that's not 0 is true. So both versions are correct, but the original removes an operation which would most likely be removed by the compiler anyway.

yyzone, 2016-06-03:
the AND was added to ignore the W component of SIMD vector. 
if we assume that all 3 vectors of both matrices have W set to 0 then we'd need to have:
return (0xF == _mm_movemask_ps((__m128)c0));
to give us correct result.
Or we AND with 0x7 and then check that the result is equal to 0x7 to safely ignore the W is 0 assumption:
int m = _mm_movemask_ps((__m128)c0);
return (0x7 == (m & 0x7));
does that make sense?

erwincoumans, 2016-06-04:
Yes, this was a bug indeed. The btMatrix3x3 == operator isn't used in Bullet, neither is the btTransform == operator (there was one spot in btCompoundShape.h which refers to the operator, but that was dead code too).
Let's fix it, thanks for the contribution!

erwincoumans, 2016-06-04:
Will be fixed here:
https://github.com/bulletphysics/bullet3/pull/647

closed 2016-06-04 by erwincoumans

yyzone, 2016-06-04:
glad to help, saludos!