bulletphysics/bullet3 #555: Make use of SYCL instead of openCL ?

LifeIsStrange, 2016-01-02:
Hello sorry I say something of stupid but there is this New khronos specification : SYCL and it aims to be the same things than opencl but using c++, this api is highter level and more easy. 

See: https://www.khronos.org/registry/sycl/specs/sycl-1.2.pdf
And : https://github.com/amd/triSYCL

erwincoumans, 2019-12-18:
closing out-of-date issues, OpenCL is not supported.

closed 2019-12-18 by erwincoumans

LifeIsStrange, 2019-12-18:
Sorry but what happened to the openCL bullet 3.X port?

erwincoumans, 2019-12-18:
Nothing. My focus is on PyBullet and reinforcement learning for robotics nowadays. Noone works on OpenCL.

LifeIsStrange, 2019-12-18:
OK !
But I believe that openCL is a way to get more performance, as such it could be useful for reinforcement learning too. 

But yes nobody is working on it for now.