bulletphysics/bullet3 #556: Release date of bullet 3.0 ?

LifeIsStrange, 2016-01-02:
Hello, Sorry to take your time but I will create a video game and I will use Bullet 3.0 if it is released in 2016, if not I will use havok or others. 
So do you think that bullet 3.0 with openCL will be released in 2016 ? :)

btrace, 2016-02-11:
this question is also what i want to know

erwincoumans, 2016-02-11:
The OpenCL implementation is available in experimental state. You can use the --enable_experimental_opencl command-line flag to try out a few examples. There is no active OpenCL/GPU development and there have been no contributions for the OpenCL/GPU work so there is no roadmap for that part.

LifeIsStrange, 2016-02-11:
Thanks, I think you should ask help to AMD, since gpuopen (http://wccftech.com/amds-answer-to-nvidias-gameworks-gpuopen-announced-open-source-tools-graphics-effects-and-libraries/)  they need for be on par with nvidia gameworks an open alternative to physx. 

The openCL support is no longer a problem, opencl 1.2 is supported on all pc drivers on all platforms, even on Android. More and more mainstream softs use opencl, like Gimp,photoshop , libreoffice, blender, civilisation 5, etc.

LifeIsStrange, 2016-02-11:
Also from what I know you work at Google ? 
This could be great to have a powerfull physic engine on Android

closed 2021-04-26 by erwincoumans