bulletphysics/bullet3 #558: btQuantizedBvh::setQuantizationValues issue

velkyel, 2016-01-08:
I have compile my game with clang sanitizers (address, return, null, alignment, bool, bounds, enum, float-cast-overflow, float-divide-by-zero, integer-divide-by-zero, object-size, unreachable, vla-bound) and found quite strange code in btQuantizedBvh::setQuantizationValues().
Sanitizer stops at btQuantizedBvh::quantize():

    out[0] = (unsigned short) (((unsigned short)(v.getX()+btScalar(1.)) | 1));

v.getX() is out of unsigned short range(!).  It's call from setQuantizationValues():  quantize(vecIn, m_bvhAabbMax, true);  I think m_bvhQuantization should be recalculated immediately after requantization m_bvhAabbMin.  So:

    {
      unsigned short vecIn[3];
      btVector3 v;
      {
        quantize(vecIn, m_bvhAabbMin, false);
        v = unQuantize(vecIn);
        m_bvhAabbMin.setMin(v - clampValue);
      }
      aabbSize = m_bvhAabbMax - m_bvhAabbMin;                  // new
      m_bvhQuantization = btVector3(65533.0f) / aabbSize;      // new
      {
        quantize(vecIn, m_bvhAabbMax, true);
        v = unQuantize(vecIn);
        m_bvhAabbMax.setMax(v + clampValue);
      }
      aabbSize = m_bvhAabbMax - m_bvhAabbMin;
      m_bvhQuantization = btVector3(65533.0f) / aabbSize;
    }

Anyway.  Are you sure that these requantizations of m_bvhAabbMin/Max are necessary?

erwincoumans, 2016-02-23:
Thanks for the feedback, I'll look into it. In general, re-quantization is needed if/when the world AABB size changes.

erwincoumans, 2016-02-23:
It should be fixed in this pull request:
https://github.com/bulletphysics/bullet3/pull/570

closed 2016-02-23 by velkyel