bulletphysics/bullet3 #559: Build fails when BUILD_EXTRAS=OFF

scpeters, 2016-01-12:
`Test_BulletInverseDynamics` fails to build when `BUILD_EXTRAS=OFF`, since the `BulletInverseDynamicsUtils` library can't be found.

After configuring with:

~~~
cmake .. -DBUILD_EXTRAS=OFF
~~~

I get the following error after calling `make`:

~~~
ld: library not found for -lBulletInverseDynamicsUtils
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test/InverseDynamics/Test_BulletInverseDynamics] Error 1
make[1]: *** [test/InverseDynamics/CMakeFiles/Test_BulletInverseDynamics.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
~~~

scpeters, 2016-01-12:
I tried disabling that test with the following:

~~~
diff --git a/test/CMakeLists.txt b/test/CMakeLists.txt
index e8b0842..f7f263e 100644
--- a/test/CMakeLists.txt
+++ b/test/CMakeLists.txt
@@ -1,6 +1,8 @@
 IF(BUILD_BULLET3)
        #SUBDIRS( TestBullet3OpenCL )
-       SUBDIRS(  InverseDynamics )
+  IF(BUILD_EXTRAS)
+         SUBDIRS(  InverseDynamics )
+  ENDIF()
 ENDIF(BUILD_BULLET3)
 
 SUBDIRS(  gtest-1.7.0  collision BulletDynamics/pendulum )
~~~

and then the `App_ExampleBrowser` fails to link:

~~~
ld: library not found for -lBulletInverseDynamicsUtils
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [examples/ExampleBrowser/App_ExampleBrowser] Error 1
make[1]: *** [examples/ExampleBrowser/CMakeFiles/App_ExampleBrowser.dir/all] Error 2
make: *** [all] Error 2
~~~

scpeters, 2016-01-12:
I found this while trying to update the bullet homebrew formula, since it turns off extras during the build by default.

erwincoumans, 2016-01-12:
Thanks for reporting, I'll look into it and fix it.

erwincoumans, 2016-01-12:
Does the homebrew formula build the example browser by default (but not the extras)?

scpeters, 2016-01-12:
The default homebrew arguments are:

~~~
cmake .. \
  -DBUILD_BULLET2_DEMOS=OFF \
  -DBUILD_EXTRAS=OFF
~~~

So that doesn't build the example browser, but it does try to build the tests.

scpeters, 2016-01-12:
I can also disable tests for the homebrew build; that should be a suitable workaround.

hasufell, 2017-05-28:
still true

closed 2021-04-26 by erwincoumans