bulletphysics/bullet3 #561: Not really an issue but did this repository win a Sci-Tech Academy Award nomination?

SocalSam, 2016-01-21:
Apologies for this not being a true issue report, but I noticed that Variety reported:

[Academy Sci-Tech Awards To Honor 21 Achievements] (http://variety.com/2015/film/awards/academy-sci-tech-awards-to-honor-21-achievements-1201403920/)

"Erwin Coumans for the development of the Bullet physics library, and to Nafees Bin Zafar and Stephen Marshall for the separate development of two large-scale destruction simulation systems based on Bullet."

executionunit, 2016-01-21:
Apologies for commenting on a false issue report but:

Congratulations to all concerned, well deserved.

erwincoumans, 2016-01-21:
Yes, Bullet has been widely used in the movie industry and we received an Academy Award for it. Let's close the issue.

closed 2016-01-21 by erwincoumans