bulletphysics/bullet3 #562: Gpu demo

jordanbonilla, 2016-01-23:
After I install this version of bullet, what examples can I run to compare running time with and without the GPU collision detection optimizations?

erwincoumans, 2016-02-11:
Most of the OpenCL/GPU demos are not enabled in the transition to the new Bullet Example Browser. You can run the example browser with the --enable_experimental_opencl command line flag and then there will be a section with a few OpenCL examples. Again, most of the samples are not ported yet. but the source code is in examples/OpenCL. If you feel like converting them, please consider contributing.

samoatesgames, 2016-02-17:
Maybe I am being dumb, but I can't get the GPU examples to show.
Looking into it, the CMakeLists looks for the '${BULLET_PHYSICS_SOURCE_DIR}/Demos3' directory otherwise it sets NO_OPENGL3.

There is no Demos3 folder though?

Flix01, 2016-02-24:
> Maybe I am being dumb, but I can't get the GPU examples to show.

I've just tried them now and they seems to work for me (box-box and pair-bench demos).
Have you read the **README.md** file ?
I suggest you use premake instead of CMake: it's precompiled in the build3 folder fo Linux, Linux64, Windows and OSX.

P.S. the only additional step I needed to make the openCL demos work on my (Ubuntu) Linux64 machine was to install the _nvidia-modprobe_ package (otherwise I need to have to be super user to be allowed to use Cuda/OpenCL in all the Linux applications in general, but this is only related to Ubuntu+Nvidia AFAIK).

> There is no Demos3 folder though? 

Nope, AFAIK they should be in the examples/OpenCL folder.

erwincoumans, 2016-02-24:
Indeed, the cmakelist files for opencl are out-of-date and need fixing, please use premake for now.

erwincoumans, 2019-12-18:
closing out-of-date issues, OpenCL is not supported.

closed 2019-12-18 by erwincoumans