bulletphysics/bullet3 #564: are there some books about this physics engine for learning ?

btrace, 2016-02-16:
hi! i am a c++ programmer and very interesting in physics engines, but not good at physics theory about how to write an engine, so it is difficult for me understanding the algorithms. are there some books about this physics engine to start? sorry for my poor english :-D

marcossuzuki, 2016-02-16:
start with the physical books from school and college.

aricd, 2016-02-16:
For a gentle introduction to mechanical physics I'd recommend "Conceptual Physics by Paul G. Hewitt".  It starts off assuming you have familiarity with algebra, not necessarily calculus as found in University level books.

Besides books I'd recommend that you implement some basics on your own starting with a 2D point on the screen and:
- applying a set velocity to the point to make it move on the screen
- bound the point by the screen, and when it collides make it's velocity reflect appropriately
- use keyboard input to apply velocities to the point (up, down, left, right)
- apply gravity to the point
- make a point have a radius, and have collision react appropriately
- make many points, and have them collide, and bounce correctly
- see how many points you can get on a screen at a decent frame-rate, then try to optimize your code!

This may be trivial for you, or it may be a significant amount of effort, but this will give you some insight into the basic concerns of a physics engine. 

Afterwards, I'd recommend using bullet, or box2D, to make a simple game that duplicates what you've created, and and stepping into some of the code to get an idea of how those engines implemented.

-Good Luck!

CGAnderson, 2016-02-16:
You could also try:
Game Physics Engine Development by Ian Millington (older editions ok, too)
and
Learning Game Physics with Bullet Physics and OpenGL by Chris Dickinson  (not current Bullet though).

btrace, 2016-02-21:
thanks for all detailed and helpful suggests！

benelot, 2016-05-03:
Be sure to close this issue, otherwise we produce clutter in the issuetracker. Your question and further questions should better go into the Forum on the Bullet Physics website.

closed 2017-09-23 by btrace