bulletphysics/bullet3 #582: No Collision between constrained rigidbodies after constraint has broken (Blender)

scorpion81, 2016-03-11:
This potential bug in bullet occurs in latest blender 2.77 buildbot version and has been reported in blenders bugtracker as well, but the feedback (via IRC) was that bullet might  be the cause. Especially a couple of changes in sequential impulse constraint solver between 2.82 and 2.83 could be the culprit.
Objects still penetrate each other after the constraint has broken IF disableLinkedCollision is disabled.
Works without disableLinkedCollision, but this yields different simulation behavior in general.
So far only the viewport integration of bullet into blender was tested, not the game engine.

See here also https://developer.blender.org/T47758 for more details and a testfile.

brechtvl, 2016-03-27:
It seems this is a bug in Bullet, introduced in #165 . The previous code would check if the constraint is enabled before disabling collisions, the new code no longer does that as far as I can tell.

erwincoumans, 2016-03-27:
Thanks for the report Brecht, I'll look into fixing it soon.

closed 2021-04-26 by erwincoumans