bulletphysics/bullet3 #588: Compound collisions: Accessing the btCollisionShape pair for a contact point.

kermado, 2016-04-01:
The `btManifoldPoint` class has the fields `m_index0` and `m_index1` that I thought represented the index of the `btCollisionShape` in the `btCompoundShape` that collided. Unfortunately, this won't work for recursive compound shapes and it doesn't appear to work for gImpact trimesh shapes either.

The `btGImpactCollisionAlgorithm::addContactPoint` function does the following:

```
m_resultOut->setShapeIdentifiersA(m_part0,m_triface0);
```

Which means that I get what I guess is the index of the trimesh face that collided, rather than the index of the gImpact shape in the `btCompound` shape.

Can we add fields to `btManifoldPoint` that point to the pair of collision shapes?

There is some discussion related to this here: http://www.bulletphysics.org/Bullet/phpBB3/viewtopic.php?f=9&t=11056

cameronwhite, 2016-04-01:
If you have access to the `btCollisionObjectWrapper`, such as in the callback when a contact is added, you can walk up the hierarchy to find that information. For example:

```cpp
int findCompoundShapeIndex(const btCollisionObjectWrapper *obj)
{
    while (obj->m_parent)
    {
        if (obj->m_parent->getCollisionShape()->isCompound())
            return obj->m_index;
        else
            obj = obj->m_parent;
    }

    return -1;
}
```

kermado, 2016-04-04:
Thanks, but I'm looking to find the pair of **non-compound** collision shapes that collided. I note that the `btCollisionObjectWrapper::getCollisionShape()` function exists. However, it appears that this doesn't work correctly for concave gImpact mesh shapes inside of compound shapes. In such cases `getUserPointer` is returning 0 when I have definitely set the user pointer for the gImpact shape.

kermado, 2016-04-05:
So it looks like when gImpact creates the `btCollisionObjectWrapper` it provides a `btTriangleShapeEx` and not the `btGImpactMeshShape`. In other words, gImpact reports the triangle that collided rather than the mesh shape. This makes sense, but I needed the user pointer for the `btGImpactMeshShape`.

For now I'm just setting the user pointer for the mesh part to be equal to the user pointer for the mesh shape and the user pointer for the triangle to be equal to the user pointer for the mesh part in `btGImpactCollisionAlgorithm::gimpact_vs_shape`.

closed 2016-04-05 by kermado

erwincoumans, 2016-04-05:
the m_collisionShape in the collisionObjectWrapper still points to you gimpact shape, no need for custom pointers

kermado, 2016-04-11:
I don't see any `m_collisionShape` member of `btCollisionObjectWrapper`. There only looks to be `m_shape`, which is exactly what `getCollisionShape` returns.

erwincoumans, 2016-04-11:
I meant, use the collision shape of the m_collisionObject, which is stored in the btCollisionObjectWrapper, that should point to the original gimpact shape:

objectWrap->getCollisionObject()->getCollisionShape();

kermado, 2016-04-25:
That won't work for gImpact shapes that form part of a compound hierarchy more than one level deep. In such cases `getCollisionShape()` will return the compound collision shape and the `m_index0` and the `m_index1` fields are not sufficient to determine which children were actually involved in the collision.