bulletphysics/bullet3 #593: Getting torque applied by enableAngularMotor

Gregwar, 2016-04-11:
Hello,
When calling `enableAngularMotor` on an hinge joint, there is no way to get the impulse that is effectively applied on the joint. `getAppliedImpulse` returns an impulse that was used for both the motor and the 
How this could be achieved?
Also, what is the policy of `enableAngularMotor` to reach the given speed?

Gregwar, 2016-04-11:
Note that `getJointFeedback()` returns NULL even when calling `enableFeedback(true)`

Gregwar, 2016-04-11:
I eventually tried:

```c++
btJointFeedback *fb = new btJointFeedback();
hinge->setJointFeedback(fb);
```

And then later have a look at the data populated in the joint feedback
However, there is still applied torque even if I set the maximum impulse to 0, so this is not really the information I want, any idea?

benelot, 2016-05-03:
Can you close this issue and post your question again on the bullet physics forum? We are ready to help you there, here you only get limited attention sometimes!

closed 2021-04-26 by erwincoumans