bulletphysics/bullet3 #596: convexSweepTest and ClosestConvexResultCallback return errant hitPointWorld

ktfh, 2016-04-13:
I am using btDynamicsWorld->convexSweepTest with a btSphereShape on a few static collision objects with btBoxShape. The ClosestConvexResultCallback returns reasonable values when the boxs are axis aligned but if i rotate the box on multiple axis eg x=-8deg, y=70deg, z=0deg. ClosestConvexResultCallback->m_hitPointWorld returns a point far outside the volume of the convexSweepTest but not consistently! and doing the convexSweepTest on different portions of the box changes the frequency of errant hit point values.
heres an example of a logged bad value:
`CONVEXSWEEP`
`FROM -27.449842,-2.994380,109.886520`
`TO   -27.529308,-3.604819,109.553276`
`RADIUS 0.450000`
`HITPOS -20.280796,-2.621698,101.837120`
`HITNORM -0.130782,0.990269,-0.047573`
`HITOBJ 0x1aa9510`
`HALFEXTENTS 19.919998,0.920000,49.919998`
`HITOBJPOS 0.000000,0.000000,120.000000`
`HITOBJROT -0.057141,0.572179,0.040011,0.817157`
`DISTANCE 10.785498 > 1.150001 EXPECTED`

I then tried to reproduce with a more minimal setup of just a single cube and sphere convexSweepTest using the logged from & to and box orientation and size values but it seems to return only correct values.

`FROM -27.449842,-2.994380,109.886520`
`TO   -27.529308,-3.604819,109.553276`
`RADIUS 0.450000`
`HITPOS -27.425507,-3.178623,109.895378`
`HITNORM -0.130799,0.990266,-0.047596`
`HALFEXTENTS 19.879997,0.880000,49.879997`
`HITOBJPOS 0.000000,0.000000,120.000000`
`HITOBJROT -0.057141,0.572179,0.040011,0.817157`
`DISTANCE 0.186054`

I ran with valgrind and it didn't pickup any out of bounds access

I tried compiling with g++ and clang++ on linux and mingw on windows and all had same behavior though the frequency of
bad values seemed highest with g++ and statically linking bullet, oddly.

I am guessing this is some kind of unaligned memory problem? I am not sure what my next step should be to debug this and would appreciate any advice.

heres an image
[image]

ktfh, 2016-04-14:
heres a video showing the behavior [https://www.youtube.com/watch?v=tqga2pD-3pE](url)

ktfh, 2016-04-14:
Ive made a 150 line gist that reliably reproduces the bug for me. 
[https://gist.github.com/ktfh/dc9c4d26ed2e59ef4eb43aff9a98f1cd](https://gist.github.com/ktfh/dc9c4d26ed2e59ef4eb43aff9a98f1cd)

ktfh, 2016-04-14:
heres example output 
`FRAME 162`
`EXPECTED MAX DISTANCE OF 2.450000 HITDIST 3.201911`
`FROM -5.027759,-3.729265,-4.562845`
`TO   -5.027759,-5.729265,-4.562845`
`RADIUS 0.450000`
`HITPOS -2.689933,-5.647984,-5.614190`
`HITNORM -0.130775,0.990269,-0.047586`

`FRAME 351`
`EXPECTED MAX DISTANCE OF 2.450000 HITDIST 22.545046`
`FROM -31.200333,-8.393215,-13.670421`
`TO   -31.200333,-10.393215,-13.670421`
`RADIUS 0.450000`
`HITPOS -10.807246,-7.574422,-23.248037`
`HITNORM -0.130784,0.990268,-0.047581`

`FRAME 523`
`EXPECTED MAX DISTANCE OF 2.450001 HITDIST 58.432961`
`FROM -73.791763,-14.315394,-20.716990`
`TO   -73.791763,-16.315395,-20.716990`
`RADIUS 0.450000`
`HITPOS -21.078253,-10.003813,-45.558628`
`HITNORM -0.130766,0.990271,-0.047580`

ktfh, 2016-04-15:
I started digging through the executing code during a convexSweepTest, it seems like ultimately btGjkPairDetector::getClosestPointsNonVirtual() gets called twice to determine the hit point, once with the FROM&TO positions of the convexSweepTest and a second time with some interpolated positions. I noticed during sweeps that return errant values the first call of getClosestPointNonVirtual seems to calculate a hit point close to correct but during the 2nd call of getClosestPointsNonVirtual is when an obviously wrong point is calculated.

Inside getClosestPointsNonVirtual() is a for loop with many exit if statements, it seems to always exit at btGjkPairDetector.cpp:167 when i get correct results:
`if (m_simplexSolver->inSimplex(w)) {`
`m_degenerateSimplex = 10;`
`checkSimplex=true;`
`break;}`

and on convexSweepTests where i get errant values its exiting the for loop at btGjkPairDetector.cpp:227:
`if (previousSquaredDistance - squaredDistance <= SIMD_EPSILON * previousSquaredDistance)`
`{`
`checkSimplex = true;`
`m_degenerateSimplex = 12;`
`break;`
`}`

erwincoumans, 2016-04-15:
Thanks for the report, repro case and for looking into this. I have a lot of other things on my plate, so can't quickly look into this. Could you try to comment-out the entire termination condition (if (previousSquaredDistance...) { ...}  at btGjkPairDetector.cpp:227, and see if that help you?

ktfh, 2016-04-15:
IT WORKS !

ktfh, 2016-04-15:
thanks for your help, Its a relief to see the code work again. If theres anything else I can do let me know.

ktfh, 2016-04-17:
After more fiddling with the commented out exit condition, I am noticing the convexSweepTest hit point is occasionally getting misplaced slightly inside the collision object and the hit normal gets inverted, the world position of the hit point is always within the volume of the convex sweep though instead of 40+ units away. I recorded a video to help visualize https://www.youtube.com/watch?v=v3scg_8aN38

The key ingredients to reproducing the behavior seems to be having a btRigidBody with some forces applied to it near or inside the convexSweepTest volume and a box shape collision object rotated on multiple axis getting hit by the convexSweepTest.

I am assuming the exit condition i commented out is probably necessary for some collision edge case? and probably not the actual bug but reacting adversely to it. I am going to keep digging through the code and report back if i can find anything.

rebirth123, 2016-08-17:
I'm also having a similar problem with the convex sweep test.  It occasionally gives me a collision that appears to be offset incorrectly (from the normal) and the faster (greater distance between start and end) the ghost object the greater this offset can be.  It jumps much like ktfh has described and is too unpredictable to figure out a work around.  At first I thought it was a tunnelling problem, but after closer inspection I can see the collision sphere shape going up so high that my high speed 'thing' ends up missing the barrier.  My collision checking if happening within a loop, much like the the character controller class (it's a stripped down version).

ktfh, 2016-08-17:
I found that convex sweeps against a btBvhTriangleMeshShape work reliably and put off trying to resolve this. I think the problem starts in btGjkPairDetector.cpp, something to do with bad squaredDistance value, the seperatingAxis or the simplexSolver. Comments in the code reference this https://code.google.com/archive/p/bullet/issues/250 a happy giraffe and massive panda with identical problems, their solution is a degenerate case detection but i don't think it ever executes.

erwincoumans, 2016-08-17:
>>a happy giraffe and massive panda

Haha, I'll have to look at those and fix things :-)

ktfh, 2016-08-17:
Here i wrote a simpler test that reproduces the behavior https://gist.github.com/ktfh/20a507e68b1effb8774298b9abcc6222

closed 2021-04-26 by erwincoumans