bulletphysics/bullet3 #617: Fix path on headers.

mrseth, 2016-05-05:
Installing from source on linux, the headers are installed on /usr/local/include/bullet.

However, headers don't include other headers using bullet/*, they include other headers assuming the path being provided is /usr/local/include/bullet.

For example: on BulletCollision/CollisionDispatch/btCollisionWorld.h, it includes LinearMath/btVector3.h.

But the makefile didn't put the LinearMath directory under /usr/local/include, it put it under /usr/local/include/bullet, so it can't find it.

So what people do is to use -I/usr/local/include/bullet to make it use that directory to look for stuff.

But the thing is, if I were to install bullet into another path, such as /usr/include, which is what a packager manager does, my makefile wouldn't work.

So, what I propose is to make the headers reference instead of LinearMath/btVector3.h, bullet/LinearMath/btVector3.h.

Now if the headers are installed anywhere on the system where its a valid path for looking for headers, the makefile works, no need to add a flag just to compensate for the wrong path on the header.

erwincoumans, 2016-05-05:
It is a hairy issue. Bullet consist of several libraries, such as BulletCollision, BulletDynamics and LinearMath. Ideally there would be no 'bullet' in the package managers, but directly the individual libraries. Unfortunately, LinearMath is a really bad name. I will fix that for Bullet3Common, but migrating to a new version of Bullet is hard. I prefer local Bullet installations myself, and not depend on system libraries/system install, so others contributed the system installation.
For a future version of Bullet, I would like to see:

/usr/local/include/Bullet3Common
/usr/local/include/Bullet3Collision
/usr/local/include/BulletDynamics

mrseth, 2016-05-05:
>I will fix that for Bullet3Common
The LinearMath name or the headers?

erwincoumans, 2016-05-05:
The name. Bullet is not a single library under the 'bullet' folder. I think it is better to distribute them as individual libraries, possibly grouped through dependencies. So if you would do
sudo apt-get install Bullet3Dynamics you get also Bullet3Collision and Bullet3Common.

Would that solve your issue?

mrseth, 2016-05-05:
Only if the install were to install all of them DIRECTLY under /usr/local/include instead of /usr/local/include/bullet.

The problem here is makefile installing on a location and the headers looking into a different one.

erwincoumans, 2016-05-05:
Yes, that is what I mentioned indeed, right? I'll repeat it, because you may have missed it:

/usr/local/include/Bullet3Common
/usr/local/include/Bullet3Collision
/usr/local/include/Bullet3Dynamics

mrseth, 2016-05-05:
Basically, if I am FORCED to use -I to indicate where to find a library already installed on a location where the system looks for by default, is no good.

closed 2021-04-26 by erwincoumans