bulletphysics/bullet3 #626: Cflags in bullet.pc are incorrect when built with CMAKE_INSTALL_PREFIX

simotek, 2016-05-13:
`bullet.pc.in` contains 

     Cflags: @BULLET_DOUBLE_DEF@ -I@CMAKE_INSTALL_PREFIX@/@INCLUDE_INSTALL_DIR@ -I@CMAKE_INSTALL_PREFIX@/include

while `BulletConfig.cmake.in` contains

    set ( BULLET_INCLUDE_DIR  "@INCLUDE_INSTALL_DIR@" )
    set ( BULLET_INCLUDE_DIRS "@INCLUDE_INSTALL_DIR@" )

when building with the following flags taken from the openSUSE build

    /usr/bin/cmake .. -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DCMAKE_INSTALL_LIBDIR:PATH=/usr/lib64 -DCMAKE_BUILD_TYPE=None '-DCMAKE_C_FLAGS=-fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -DNDEBUG' '-DCMAKE_CXX_FLAGS=-fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -DNDEBUG' '-DCMAKE_Fortran_FLAGS=-fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -DNDEBUG' '-DCMAKE_EXE_LINKER_FLAGS=-Wl,--as-needed -Wl,--no-undefined -Wl,-z,now' '-DCMAKE_MODULE_LINKER_FLAGS=-Wl,--as-needed -Wl,--no-undefined -Wl,-z,now' '-DCMAKE_SHARED_LINKER_FLAGS=-Wl,--as-needed -Wl,--no-undefined -Wl,-z,now' -DLIB_SUFFIX=64 -DCMAKE_SKIP_RPATH:BOOL=ON -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_STATIC_LIBS:BOOL=OFF -DCMAKE_COLOR_MAKEFILE:BOOL=OFF -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_MODULES_INSTALL_DIR=/usr/share/cmake/Modules -DCMAKE_BUILD_TYPE=RelWithDebInfo -DINCLUDE_INSTALL_DIR=/usr/include/bullet -DBUILD_BULLET2_DEMOS=OFF -DBUILD_OPENGL3_DEMOS=OFF

Particularly the combination of `-DCMAKE_INSTALL_PREFIX:PATH=/usr` and `-DINCLUDE_INSTALL_DIR=/usr/include/bullet` this leads to the CFlags in the bullet.pc file being `/usr/usr/include/bullet`.
To be consistent with what is in the  `BulletConfig.cmake.in` file `bullet.pc.in` should contain `Cflags: @BULLET_DOUBLE_DEF@ -I@INCLUDE_INSTALL_DIR@ -I@CMAKE_INSTALL_PREFIX@/include` Instead

Note that cmake is called by a RPM macro hence the extra useless flags and make install is called as `/usr/bin/make VERBOSE=1 DESTDIR=/home/abuild/rpmbuild/BUILDROOT/libbullet-2.83.7-8.1.x86_64 install/fast -C build`

erwincoumans, 2016-05-13:
Do you have a suggested pull request?

erwincoumans, 2016-05-13:
fixed here:
https://github.com/bulletphysics/bullet3/pull/627
thanks again.

akien-mga, 2016-10-27:
This issue is still valid in 2.85.1, as #627 was reverted by #679 (it would have been nice to use `git revert` for that btw).

So there is still an inconsistency in system-wide installations between:
- the pkgconfig file which expects `INCLUDE_INSTALL_DIR` to be a relative path (as it is defined by default, `"include/bullet/"`)
- the `INSTALL()` calls which all expect it to be an absolute path (otherwise with the default value, headers are installed in `build/include/bullet/` instead of `CMAKE_INSTALL_PREFIX/include/bullet/`).

akien-mga, 2017-12-30:
Could this issue be reopened? As mentioned one year ago, the "fix" has been reverted, and the issue is still valid. I still need to drag the same SUSE patch in my Mageia packaging: http://svnweb.mageia.org/packages/cauldron/bullet/current/SOURCES/bullet-suse-Fix-bullet.pc.patch?view=markup

erwincoumans, 2017-12-30:
Please provide a patch and if it doesn't break the build I'll apply it.

closed 2021-04-26 by erwincoumans